LibreELEC/LibreELEC.tv #5317: [BUG] Setting the regulatory domain for WLAN doesn't work

schornstephan, 2021-04-12:
### Describe the bug
I set my regulatory domain to Germany on first run or some time after that. Now, I need to change it to US. Let's just assume I moved there. I can set the regulatory domain to US via GUI. However, "iw reg get" will still show Germany instead of the new country. I can still connect on channels I shouldn't be able to connect on. The change is correctly applied to ~/.cache/regdomain.conf.

### To Reproduce
Steps to reproduce the behavior:
1. Set new regulatory domain to US via GUI or "iw reg set US".
2. Reboot or Reconnect WLAN network or "systemctl restart wpa_supplicant".
3. Changes aren't applied, although GUI shows they were applied.

### Informations
 - LE Version: 9.2.6
 - Hardware Platform: RPi4-2GB

### Log file
dmesg ends after 6 seconds uptime. If you want to see it anyway, I'll upload it.

### Additional context
I absolutely have no idea where Libreelec is still getting that setting "Germany" from. I tried to find out for hours where it is stored besides ~/.cache/regdomain.conf. I have no idea. If you tell me where it is stored, I can fix this bug myself. Normally, it would be /etc/wpa_supplicant.conf, but this file does not exist in Libreelec.

chewitt, 2021-04-12:
Please verify the behaviour on LE10b1, the LE9.2 codebase will not receive any further updates/fixes.

schornstephan, 2021-04-12:
Sure, I'll get back to you when I've tested 10b1.

heitbaum, 2021-04-23:
@schornstephan - how did you go?

chewitt, 2021-06-13:
Closing due to lack of response. Feel free to submit the info asked for and the ticket can be reopened.

closed 2021-06-13 by chewitt