LibreELEC/LibreELEC.tv #5318: [BUG] iwd kernel warning

heitbaum, 2021-04-12:
### Describe the bug

Dmesg warning with net/wireless/nl80211.c:7260 nl80211_get_reg_do

https://bugs.gentoo.org/746539
https://bugzilla.kernel.org/show_bug.cgi?id=208599

### To Reproduce
Steps to reproduce the behavior:
1. Use iwd with LE10

### Informations
 - LE Version: LE10
 - Hardware Platform: Generic

### Log file

Refer to attached bug reports.

### Additional context

antonlacon, 2021-08-29:
https://lore.kernel.org/linux-wireless/iwlwifi.20210409123755.ba2ea961f4ae.I8fde32d3196e860efa3b4ec464c42194195b42ec@changeid/#t should be the fix. I don't believe it's been reviewed or merged yet, as I haven't located it in the mac80211.git tree on kernel.org.

heitbaum, 2021-08-29:
@antonlacon I believe the patch does indeed work. It associated to my first access point then moved straight away to the second AP.

```
wlan0     Link encap:Ethernet  HWaddr x:x:x:x:x:x
          inet addr:192.168.x.x  Bcast:192.168.x.x  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2865 errors:0 dropped:0 overruns:0 frame:0
          TX packets:450 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:739106 (721.7 KiB)  TX bytes:67747 (66.1 KiB)
```

```
[    0.000000] microcode: microcode updated early to revision 0x88, date = 2021-03-31
[    0.000000] Linux version 5.14.0-rc7 (docker@03af303e3737) (x86_64-libreelec-linux-gnu-gcc-10.3.0 (GCC) 10.3.0, GNU ld (GNU Binutils) 2.37) #1 SMP Sun Aug 29 18:18:45 AEST 2021
[    0.000000] Command line: BOOT_IMAGE=/KERNEL boot=LABEL=LIBREELEC disk=LABEL=STORAGE i915.enable_guc=2 quiet
[    3.051320] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    3.079433] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    3.104058] Intel(R) Wireless WiFi driver for Linux
[    3.104124] iwlwifi 0000:00:14.3: enabling device (0000 -> 0002)
[    3.105487] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-64.ucode failed with error -2
[    3.111922] iwlwifi 0000:00:14.3: api flags index 2 larger than supported by driver
[    3.111932] iwlwifi 0000:00:14.3: TLV_FW_FSEQ_VERSION: FSEQ Version: 89.3.35.37
[    3.112079] iwlwifi 0000:00:14.3: loaded firmware version 63.c04f3485.0 QuZ-a0-hr-b0-63.ucode op_mode iwlmvm
[    3.143452] wl: loading out-of-tree module taints kernel.
[    3.143458] wl: module license 'MIXED/Proprietary' taints kernel.
[    3.143459] Disabling lock debugging due to kernel taint
[    3.248386] iwlwifi 0000:00:14.3: Detected Intel(R) Wi-Fi 6 AX201 160MHz, REV=0x354
[    3.371347] iwlwifi 0000:00:14.3: Detected RF HR B3, rfid=0x10a100
[    3.436657] iwlwifi 0000:00:14.3: base HW address: x:x:x:x:x:x
[    4.479373] Bluetooth: Core ver 2.22
[    4.479384] NET: Registered PF_BLUETOOTH protocol family
[    4.479384] Bluetooth: HCI device and connection manager initialized
[    4.479387] Bluetooth: HCI socket layer initialized
[    4.479388] Bluetooth: L2CAP socket layer initialized
[    4.479390] Bluetooth: SCO socket layer initialized
[    4.479946] usbcore: registered new interface driver btusb
[    4.480343] Bluetooth: hci0: Bootloader revision 0.4 build 0 week 30 2018
[    4.481352] Bluetooth: hci0: Device revision is 2
[    4.481359] Bluetooth: hci0: Secure boot is enabled
[    4.481364] Bluetooth: hci0: OTP lock is enabled
[    4.481367] Bluetooth: hci0: API lock is enabled
[    4.481368] Bluetooth: hci0: Debug lock is disabled
[    4.481368] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[    4.485332] Bluetooth: hci0: Found device firmware: intel/ibt-19-0-4.sfi
[    4.493094] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.493095] Bluetooth: BNEP filters: protocol multicast
[    4.493098] Bluetooth: BNEP socket layer initialized
[    6.316066] Bluetooth: hci0: Waiting for firmware download to complete
[    6.316408] Bluetooth: hci0: Firmware loaded in 1788159 usecs
[    6.316551] Bluetooth: hci0: Waiting for device to boot
[    6.331600] Bluetooth: hci0: Device booted in 14789 usecs
[    6.343905] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-19-0-4.ddc
[    6.345622] Bluetooth: hci0: Applying Intel DDC parameters completed
[    6.348589] Bluetooth: hci0: Firmware revision 0.0 build 191 week 21 2021
[    8.120930] wlan0: authenticate with y:y:y:y:y:y
[    8.131728] wlan0: send auth to y:y:y:y:y:y (try 1/3)
[    8.148956] wlan0: authenticated
[    8.150962] wlan0: associate with y:y:y:y:y:y (try 1/3)
[    8.156346] wlan0: RX AssocResp from y:y:y:y:y:y (capab=0x11 status=0 aid=4)
[    8.158723] iwlwifi 0000:00:14.3: Got NSS = 4 - trimming to 2
[    8.162728] wlan0: associated
[    8.183406] iwlwifi 0000:00:14.3: Unhandled alg: 0x3f0707
[    8.201504] wlan0: Limiting TX power to 12 (15 - 3) dBm as advertised by y:y:y:y:y:y
[    8.201512] iwlwifi 0000:00:14.3: Got NSS = 4 - trimming to 2
[    8.256636] wlan0: disconnect from AP y:y:y:y:y:y for new auth to y:y:y:y:y:2
[    8.344092] wlan0: authenticate with y:y:y:y:y:2
[    8.346980] wlan0: send auth to y:y:y:y:y:2 (try 1/3)
[    8.374122] wlan0: authenticated
[    8.374267] wlan0: associate with y:y:y:y:y:2 (try 1/3)
[    8.379451] wlan0: RX ReassocResp from y:y:y:y:y:2 (capab=0x11 status=0 aid=2)
[    8.381877] iwlwifi 0000:00:14.3: Got NSS = 4 - trimming to 2
[    8.387197] wlan0: associated
[    8.387241] wlan0: Limiting TX power to 12 (15 - 3) dBm as advertised by y:y:y:y:y:2
[    8.387247] iwlwifi 0000:00:14.3: Got NSS = 4 - trimming to 2
[    8.406433] iwlwifi 0000:00:14.3: Unhandled alg: 0x3f0707
[   11.067069] Bluetooth: RFCOMM TTY layer initialized
[   11.067075] Bluetooth: RFCOMM socket layer initialized
[   11.067079] Bluetooth: RFCOMM ver 1.11
```

antonlacon, 2021-12-25:
An alternative fix should have merged in 5.15:
https://github.com/torvalds/linux/commit/eb09ae93dabfd0980f5226dfd0bd531eb87902f5#diff-94f96320b6c13db068025b4617b636db4cb02d6f0c03d3fe708405ec8437bb4f

heitbaum, 2021-12-26:
I have removed the patch referenced in https://github.com/LibreELEC/LibreELEC.tv/issues/5318#issuecomment-907740306  and I can confirm this is now working as expected without the `WARNING: CPU: 7 PID: 492 at net/wireless/nl80211.c:7260 nl80211_get_reg_do+0x1cd/0x1f0` in kernel 5.15.11

closed 2021-12-26 by heitbaum