LibreELEC/LibreELEC.tv #5328: [BUG] Librespot addon very stuttery on Raspberry Pi

Emgalai, 2021-04-19:
### Describe the bug
Librespot addon is very stuttery on Raspberry Pi, while librespot itself works fine if run separately from command line.

### To Reproduce
Steps to reproduce the behavior:
1. Install Librespot on LibreELEC running on RPi.
2. Connect to Librespot from Spotify Android app (or other source) and start playing something.
3. Librespot addon starts playback but is very stuttery and unresponsive. To the point that it is completely unusable.

### Informations
 - LE Version: 9.2.6
 - Librespot addon version: 9.2.0.125
 - Hardware Platform: Raspberry Pi (1) B+ (with IQaudio DAC+)

### Log file
http://ix.io/2WCk

### Additional context
The problem is within the addon, as running the librespot command (that is provided with the addon) works fine on the Raspberry Pi, even with 320 bitrate. My current workaround is to have the addon disabled and an autostart.sh script with content:
`librespot -b 320 -n LibreELEC &`

chewitt, 2021-04-19:
Please retest with LE10b1 because any investigation and fixing work will be done there, not on LE9.2.6.

chewitt, 2021-04-19:
Ahh, RPi1B .. so ignore the LE10 comment, you're stuck with LE9.2.6

heitbaum, 2023-05-02:
Closing - new version of librespot (LE12) added, but Official back port to le9 or le10 not planned.
- #7732

closed 2023-08-27 by heitbaum