LibreELEC/LibreELEC.tv #5335: [BUG] Black screen when using analog output

Hiki1, 2021-04-22:
### Describe the bug
Black screen occurs during boot, when using composite output of the Raspberry Pi

### To Reproduce
Steps to reproduce the behavior:
1. Fresh installation of 9.95.2
2. Edit the config.txt and set **enable_tvout=1** there
3. Connect the RPi to a CRT TV via analog output - jack to composite cable is required
4. Initially the colored box appear like [here]([image]) for short while
5. Black screen occurs and stays like that

### Informations
 - LE Version: 9.95.2
 - Hardware Platform: RPi4

### Additional context
Jack to RCA cable looks like [that]([image]).

HiassofT, 2021-04-22:
Initial support for tvout has landed in RPi kernel just yesterday https://github.com/raspberrypi/linux/pull/4240

These changes will be included in future LE versions (but they came too late for beta2)

Hiki1, 2021-04-22:
Sounds great! Thanks for the info!

Hiki1, 2021-05-22:
I've retested this issue on a 9.95.3 and unfortunately it still does occur.

HiassofT, 2021-05-22:
9.95.3 uses the same kernel as 9.95.2 - we had to revert the kernel bump as it caused issues and we are still waiting for a fix.

heitbaum, 2022-02-17:
@Hiki1 closing this ticket. If you are still having issue with LE10/11 - please reopen this bug.

closed 2022-02-17 by heitbaum