LibreELEC/LibreELEC.tv #5353: [BUG] Snapcast server addon not installable due to missing service.librespot (LE10)

HorstBaerbel, 2021-05-03:
### Describe the bug
Snapcast server addon not installable due to missing dependency service.librespot

### To Reproduce
Steps to reproduce the behavior:
1. Install LE10 Beta
2. Try to install Snapcast server

### Informations
 - LE Version: 10
 - Hardware Platform: amd64
 
Snapcast support was added [here](https://github.com/LibreELEC/LibreELEC.tv/pull/2597#issuecomment-376349804). Please comment if you need more info

lrusak, 2021-05-03:
Refresh the add-on repo and try again.

closed 2021-05-03 by lrusak

HorstBaerbel, 2021-05-03:
I did [this](https://wiki.libreelec.tv/how-to/force-add-on-update) and the LibreElec Addon Repo is at version 9.80.11, but nothing changed

HiassofT, 2021-05-03:
librespot is currently not available on Generic/x86 due to this bug https://github.com/LibreELEC/LibreELEC.tv/issues/4719

heitbaum, 2021-06-19:
@HorstBaerbel - draft PR raised for librespot Generic - relies on the rust update #5446