LibreELEC/LibreELEC.tv #5366: [BUG] Enable autoupdate of bootloader and via usb on every reboot

MartinX3, 2021-05-10:
### Describe the bug
Currently i can update the EEPROM-Bootloader and VIA USB 3.0 Firmware once on a reboot and need to reactivate each time.
I administrate the libre elec installations of family member and would need to manually search for updates and upgrade them everytime at their location.

I would like to have an option to autoupdate the stuff on every reboot instead only once.
The raspianOS does it automatically on every reboot.
And since have the auto updater of LE it would be nice to upgrade the other stuff on every automatical LE reboot as well.

### To Reproduce
Steps to reproduce the behavior:
1. Use the updater settings

### Informations
 - LE Version: [e.g. 9.2.1] 9.95.2
 - Hardware Platform: [e.g. RPi3] RPi4

### Log file

### Additional context

HiassofT, 2021-05-10:
Please use the forum for feature request, this tracker is only for reporting bugs.

Putting `rpi-eeprom-update -a` into `.config/autostart.sh` should to what you want.

closed 2021-05-10 by HiassofT

MartinX3, 2021-05-10:
@HiassofT 
Thank you for your help
I didn't know that
Could you put this information into the README.md and adding a issue tracker template named "Feature Request" containing big letters, that we should ask such stuff in the forum?
That would avoid wrong tickets like mine in the future. :)