LibreELEC/LibreELEC.tv #5373: [BUG] crash/restart when fastforward or fastbackward a video (both in m3u lists and plex)

RManPT, 2021-05-15:
### Describe the bug
No matter what the video source is or if played directly or using plex, after opening a file and going forward or backward causes kodi to restart (not full reboot).

### To Reproduce
Steps to reproduce the behavior:
1. Open a video (no matter if PVR channels,Media library or using Plex, also doesn't matter the video format or resolution)
2. Fastforward, sometimes it only happens when you adjust again (forward or backwards) 
3. Kodi restarts (not full reboot)

### Informations
 - LE Version: 9.95.2
 - Hardware Platform: RPI4 4GB

### Log file
I am adding the .old log file as the .log file is rewritten when kodi restarts
https://pastebin.com/tz9SKwDk

### Additional context
Have tried with DRM PRIME on/off (with hardware acceleration on/off, I manly use it off due to deficient sound quality with it on)

heitbaum, 2021-08-05:
@RManPT can you please recheck with the latest nightlife’s and let us know?

heitbaum, 2022-02-17:
Closing as no response. Please reopen if still an issue - with updated logs.

closed 2022-02-17 by heitbaum