LibreELEC/LibreELEC.tv #5382: Support Xbox One controllers wirelessly

nonobio, 2021-05-20:
### Describe the bug
My Xbox One Controllers are supported when connected with USB wire but not wirelessly with the USB wireless dongle.

I created a post on the LibreElec forum : [here](https://forum.libreelec.tv/thread/14792-xbox-wireless-adapter-for-windows-v1-and-v2-support/?postID=145985#post145985).

### To Reproduce
Steps to reproduce the behavior:
1. Plug the dongle
2. Try to connect (pair) controllers with dongle
3. It doesn't work

### Informations
 - LE Version: 9.2.6
 - Hardware Platform: x64 generic PC CPU AMD E1-1200 GPU AMD Radeon HD 7310 RAM 4G SSD 128 Go Wifi

### Log file
I uploaded some informations [here](https://forum.libreelec.tv/thread/14792-xbox-wireless-adapter-for-windows-v1-and-v2-support/?postID=113385#post113385).

### Additional context

There is a project called [xow : a Linux driver for the Xbox One Wireless dongle](https://github.com/medusalix/xow) , could it be added toLibreElec ? or is it possible to install it manually ?

Thanks

SupervisedThinking, 2021-05-20:
Well https://github.com/atar-axis/xpadneo is an option but xow not since it's based on proprietary firmware which can't be shipped by LibreELEC.

Anyway if you're into retro-gaming & if you can deal with Bluetooth you should have a look at my thread https://forum.libreelec.tv/thread/23644-libreelec-rr-10-x-brave-spotify-moonlight-emulationstation-retroarch-pegasus/?pageNo=1 & ping https://forum.libreelec.tv/core/user/1723-ramalama/ since he asked for the same.

If you're able to compile a generic image yourself you can clone my repo, enable non-free packages and you'll get an image with built in xow.

nonobio, 2021-05-20:
Thanks for your reply and links.

I'm not able to compile image myself and i would like to stay with original LibreElec, so i think i will use my USB cable ;) .

Have a nice day.

CvH, 2021-05-20:
if the driver is converted to a driver-addon its likely plug and play, sadly still not mergeable for us but you could use it without the compiling stuff