LibreELEC/LibreELEC.tv #5398: [BUG] LE settings addon causes crash when Bluetooth is updating the window

heitbaum, 2021-05-25:
Crash caused during Bluetooth updating the window in LE Settings:
- This does include "LibreELEC-settings: update to bfec474" #5384 and #5385 along with the changes in my https://github.com/heitbaum/LibreELEC.tv/tree/dev
- This occurred on the Tanix TX6

http://ix.io/3nNd

```
2021-05-23 15:34:24.405 T:692     ERROR <general>: Control 1000 in window 13001 has been asked to focus, but it can't
2021-05-23 15:34:24.406 T:795      INFO <general>: SETTINGS: run # Waiting
2021-05-23 15:34:50.948 T:692   WARNING <general>: CGUIWindowManager - HandleAction - ignoring action 92, because topmost modal dialog closing animation is running
2021-05-23 15:34:51.149 T:900     ERROR <general>: SETTINGS: bluetooth.menu_connections # AttributeError("module 'oe' has no attribute 'winOeMain'")
2021-05-23 15:34:51.150 T:900     ERROR <general>: Traceback (most recent call last):
                                                     File "/usr/share/kodi/addons/service.libreelec.settings/resources/lib/log.py", line 45, in _log_function_2
                                                     File "/usr/share/kodi/addons/service.libreelec.settings/resources/lib/modules/bluetooth.py", line 338, in menu_connections
                                                   AttributeError: module 'oe' has no attribute 'winOeMain'
```

jernejsk, 2021-06-04:
I don't think this is really Allwinner specific?

heitbaum, 2021-06-05:
No. I think it is across all. Though I haven’t tested it further on any other images.

romanvm, 2021-06-19:
@heitbaum  Could you try to reproduce this with the current master and paste the log here? It includes extended error logging that should give us more info. Please attach your log to the post or upload it to some pastebin site.

Also it would be good if you describe the steps to reproduce this.

heitbaum, 2021-06-19:
I have just done the pre testing now.

was able to get the crash before updating. I went to the BT menu in settings, and watched it refresh (jerky) I did now press in to any BT device, I then moved out of the BT menu, out of le menu and was about to press enter on the system information settings and it crashed.

LibreELEC:~/.kodi/temp # pastebinit kodi.old.log
http://ix.io/3qrx
LibreELEC:~/.kodi/temp # pastebinit kodi_crashlog_20210619125905.log
http://ix.io/3qry

Next message will have the testing after reboot into master.

heitbaum, 2021-06-20:
@romanvm Whilst I haven't received a crash.

tx6:~/.kodi/temp # pastebinit kodi.log
http://ix.io/3qvT

I did receive a hang, then it went back to the main menu - see the first image, as you can see the thread doing the update to the right pane is "locked" I did add my iPhone as a BT device.

[image]

On entering into the BT menu - It always comes up with this message before it starts searching for BT devices.

[image]

heitbaum, 2021-06-20:
The follow short video also shows that when a refresh occurs the pane on the right looses focus and takes you back to "Bluetooth" thus the "down arrow" takes you to "Services"

https://user-images.githubusercontent.com/6086324/122663718-ea13a400-d1df-11eb-833b-cbeaa41ab8a3.mp4

romanvm, 2021-06-20:
@heitbaum  From the latest log I see that you are not using the latest master of the settings addon with improved diagnostics. I really want to see extended error logs. Also this comment:

> On entering into the BT menu - It always comes up with this message before it starts searching for BT devices.

makes me thing that you are using the old version. One of my changes was to display this label only when necessary. Please make sure you are testing the latest version from here: https://github.com/LibreELEC/LibreELEC.tv/issues/4547

heitbaum, 2022-02-17:
Closing. Not seen recently.

closed 2022-02-17 by heitbaum