LibreELEC/LibreELEC.tv #5404: Syncthing Version too old, Not connecting to other devices[BUG]

Lemaya, 2021-05-27:
### Describe the bug
The Syncthing version (/packages/addons/service/syncthing) in the 9.2 repository is 1.4.2.

Due to the difference of the versions the other devices don't connect with the syncthing instance running on the Raspberry Pi.

The current Version is v1.16.1, the master branch is updated to 1.15.1

Is it possible to update the packages in the 9.2 branch?

Thank you very much.

### Informations
 - LE Version: [e.g. 9.2.1]
 - Hardware Platform: Raspberry 3+B

heitbaum, 2021-08-05:
I have had a quick look and any greater than 1.4.2 fails compile using 9.2 with “cannot load crypto/ed25519: cannot find module providing package crypto/ed25519” - this is due to golang 1.13 required. Checking options to workaround this.

heitbaum, 2021-08-05:
No sensible workaround.

- 1.5.0 <-> 1.8.0 requires golang 1.13.x
- 1.9.0 >= requires golang 1.14.x

golang is used by both syncthing and docker (runc,containerd,libnetwork)

draft PR created - #5531

heitbaum, 2021-10-09:
@Lemaya - were you able to test this?

Lemaya, 2021-10-11:
No, I've been running the Docker version: https://hub.docker.com/r/linuxserver/syncthing since then. That has been working fine.

heitbaum, 2021-10-11:
perfect - I'll close this as "worked around"

closed 2021-10-11 by heitbaum