LibreELEC/LibreELEC.tv #5410: Servus ...... Today I made it to update my RPI 3 to 9.2.7. Got the stupid feeling that something went wrong with the versions. I even put on a fresh SD card because I thought it would be useful. I was able to install a few add-ons, and the Widvine was also updated to the latest version. After a while I lose the internet connection, regardless of whether it is cable or wifi .... Joyn and TVnow give me an error message. I can't even get into the Libreelec Configuration. And if I manage to do it after umpteen reboots and go for updates, the options that I have in the RPI 4 settings are offered to update the firmware of the Raspberry ... In other words, the EEPRoom and the USB bootloader function. I strongly suspect that the 9.2.7 for 18.9 (RPI 3) is still faulty, because the firmware update function actually has no place in the RPI 3. With the RPI 4 variant everything went so far yesterday evening. I'm not getting any further with the facility at the moment .....[BUG]

Oldschooler76, 2021-05-31:
### Describe the bug

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Play '....'
3. See error

### Informations
 - LE Version: [e.g. 9.2.1]
 - Hardware Platform: [e.g. RPi3]

### Log file

### Additional context

Oldschooler76, 2021-05-31:
Hello my dear ones ...... Today I made it to update my RPI 3 to 9.2.7. Got the stupid feeling that something went wrong with the versions. I even put on a fresh SD card because I thought it would be useful. I was able to install a few add-ons, and the Widvine was also updated to the latest version. After a while I lose the internet connection, regardless of whether it is cable or wifi .... Joyn and TVnow give me an error message. I can't even get into the Libreelec Configuration. And if I manage to do it after umpteen reboots and go for updates, the options that I have in the RPI 4 settings are offered to update the firmware of the Raspberry ... In other words, the EEPRoom and the USB bootloader function. I strongly suspect that the 9.2.7 for 18.9 (RPI 3) is still faulty, because the firmware update function actually has no place in the RPI 3. With the RPI 4 variant everything went so far yesterday evening. I'm not getting any further with the facility at the moment.

lrusak, 2021-05-31:
Please use the forum for support. Thanks! https://forum.libreelec.tv/

closed 2021-05-31 by lrusak