LibreELEC/LibreELEC.tv #5421: [BUG] ROC-RK3328-CC freezes randomly after some minutes

Cavallium, 2021-06-05:
### Describe the bug
After some minutes playing videos the sbc freezes completely (not only kodi, the entire os).
Before freezing, dmesg prints an error message, then just before freezing it prints some error messages regarding stalled CPUs.

### To Reproduce
Steps to reproduce the behavior:
1. Turn on the sbc
2. Open some videos (I don't know if it happens also without doing anything)
3. Wait some minutes (10 minutes to more than an hour)
4. dmesg prints an error
5. Wait some minutes (1 to 5 minutes)
6. dmesg print stalled CPU warnings
7. LibreElec freezes
8. Now the SBC can only be restarted removing the power supply.

### Informations
 - LE Version: LibreELEC-RK3328.arm-9.95.4-roc-cc
 - Hardware Platform: ROC-RK3328-CC (4GiB variant, with 128GiB eMMC)
 - Plugins/services installed:
   - tvheadend client
   - tvheadend server

### Log file
http://sprunge.us/s2CuIZ

### Additional context
Please note that the first error timestamp is many minutes ahead of the full halt of the os.

knaerzche, 2021-06-06:
Thanks for your report.
Could you please download the latest image https://test.libreelec.tv/ for **rock64** and flash it to a spare SD card. After that please copy the attached dtb file (unzip it first) 
[rk3328-roc-cc.zip](https://github.com/LibreELEC/LibreELEC.tv/files/6603870/rk3328-roc-cc.zip)
to LIBREELEC partition, edit extlinux/extlinux conf and replace `rk3328-rock64.dtb` with `rk3328-roc-cc.dtb` and test if it still 
Thanks.

Cavallium, 2021-06-06:
I turned it on an hour ago and it seems to work fine.
I'll keep it on for some hours to see if it occurs again.

Cavallium, 2021-06-07:
It works perfectly with that dtb file, it's still working since yesterday.

knaerzche, 2021-06-07:
The fix will be included after https://github.com/LibreELEC/LibreELEC.tv/pull/5418 is merged.

closed 2021-06-07 by knaerzche

Cavallium, 2021-06-07:
I noticed that using this image+dtb the 4k resolution (3840x2160) shows only the first quarter of the display stretched on the entire display.
This only happens with 3840x2160, while 4096x2160 and all other resolutions are working correctly.

knaerzche, 2021-06-07:
Actually this device tree wasn't changed - I just wanted you to take the rock64 image.
Please re-check with next nighlies - presumably available tomorrow.