LibreELEC/LibreELEC.tv #5425: Wi Fi support for OrangePi Plus 2

FSTK-Development, 2021-06-06:
Dear developers, please add Realtek RTL8189ETV support for OrangePi Plus 2 in the next release. My attempts to use LibreELEC failed. The Internet via Ethernet and Wi-Fi does not work. I will be very grateful! Here is the link to the driver: https://github.com/vpei/rtl8189etv.git

FSTK-Development, 2021-06-06:
Judging by the data linux-sunxi.org the OrangePi Plus 2 also has a Realtek RTL8211E chip for Ethernet connectivity: https://linux-sunxi.org/Ethernet#Realtek_RTL8211E

FSTK-Development, 2021-06-06:
Unfortunately, LibreELEC is absolutely useless on OrangePi Plus 2 without the internet. It is impossible to do anything, even choose your own language. So I hope for your help. Thank you for your attention!

heitbaum, 2021-06-07:
Hi @FSTK-Development  - Please use the forum for support. Thanks! https://forum.libreelec.tv/

Can you please confirm that this is your device: https://linux-sunxi.org/Xunlong_Orange_Pi_Plus_2
If so I a fix has been implemented: #5415 / #5419 and will be in the test images shortly: https://test.libreelec.tv/

With regard to your WiFi request - no additional WiFi out-of-tree drivers are being added to LE. Many references in the forum. Please request driver developer / Realtek to upstream their driver into mainline and then it can be supported.

closed 2021-06-07 by heitbaum

FSTK-Development, 2021-06-07:
I can confirm that. Upload photos and logs. For example, in the same armbian it works. Do you suggest installing the distribution yourself and installing Kodi?

heitbaum, 2021-06-07:
The Ethernet driver should work with your tv box with the next nightly then 👍

the wireless driver may have been included in Armenian. Some reading on the status and WIP for Realtek mainline as FYI. 
- https://linux-sunxi.org/Wifi#SDIO-based 
- https://wireless.wiki.kernel.org/en/users/drivers/rtl819x

FSTK-Development, 2021-06-10:
You say that the ethernet cable should work. And where to get the firmware for orangepi plus 2? Collect from the source code on github or download from the site?

heitbaum, 2021-06-10:
I believe this is the image you require.

https://test.libreelec.tv/LibreELEC-H3.arm-10.0-nightly-20210610-f547d15-orangepi-plus.img.gz

https://wiki.libreelec.tv/hardware/allwinner

FSTK-Development, 2021-06-10:
Благодарю!

FSTK-Development, 2021-06-11:
https://test.libreelec.tv/LibreELEC-H3.arm-10.0-nightly-20210610-f547d15-orangepi-plus.img.gz
I can't download it, an error is displayed:
Error.
404
I can't find what you are looking for...

jernejsk, 2021-06-11:
Pick latest image from https://test.libreelec.tv/ and ethernet should work.