LibreELEC/LibreELEC.tv #5437: [BUG] TM1623 LED display not supported by openvfd

danboid, 2021-06-14:
### Describe the bug

I booted LibreELEC-AMLGX.arm-10.0-nightly-20210612-33895da-box.img on my X96 Air Q1000 using the meson-sm1-x96-air-1000.dtb but my LED display doesn't work.

### To Reproduce
Write LibreELEC-AMLGX.arm-10.0-nightly-20210612-33895da-box.img
Adjust boot file to use meson-sm1-x96-air-1000.dtb
Insert card into X96 Air and power on.

### Informations
 - LE Version: [e.g. 9.2.1]
 10 nightly
 
 - Hardware Platform: [e.g. RPi3]
 
X96 Air Q1000 / sm1_ac213_4g

### Additional context

From inspecting the dmesg of Android on my X96A, I can see that it is using a TM1623 VFD controller. openvfd supports the TM1618 but it doesn't currently seem to support the TM1623. Looking at the vfd source, I can see that the TM1618 controller is actually just using the same controller code as the FD628 controller so I'm curious if we'll just be able to get away with adding a new enum for the TM1623 and get it to do the same?

Here is the datasheet (in Chinese) for the TM1623:

https://datasheetspdf.com/pdf-file/712580/TitanMicro/TM1623/1

I dumped and decopiled the dtb for my X96A and this is the dts code for the VFD:

```
meson-vfd {
		compatible = "amlogic,aml_vfd";
		dev_name = "meson-vfd";
		vfd_data_gpio = <0x18 0x40 0x00>;
		vfd_clock_gpio = <0x18 0x41 0x00>;
		vfd_stb_gpio = <0x6b 0x0a 0x00>;
		status = "okay";
	};
```

danboid, 2021-06-14:
Do you think the FD628 controller could work for TM1623 @chewitt ?

chewitt, 2021-06-14:
I'm closing the ticket because we exorcised openvfd from our buildsystem some time ago, thus the software we don't have cannot have a bug in hardware support. The LE openvfd repo you found is abandoned. It was forked from another repo which is still under development, but the "upstream" code has been deliberately relicensed GPLv3 which makes the kernel driver component [license] incompatible with a GPLv2 kernel. There are some other attempts at VFD support in the Allwinner and Rockchip communities which might result in some simple and upstreamable code we can use in the future, but I wouldn't hold breath.

closed 2021-06-14 by chewitt

danboid, 2021-06-15:
OK, thanks @chewitt! I am taking my config woes upstream.