LibreELEC/LibreELEC.tv #5440: [BUG] LibreELEC Master Build Error with U-boot on RK3399

mo123, 2021-06-16:
### Describe the bug

### To Reproduce
Steps to reproduce the behavior:
1. Build any RK3399 image

### Informations
 - LE Version: [e.g. 9.2.1] Master
 - Hardware Platform: [e.g. RPi3] RK3399

### Log file

https://pastebin.com/i9tMXqE1

### Additional context

lrusak, 2021-06-16:
After a quick look it almost look like something is picked up from your host system.

jernejsk, 2021-06-16:
It would be helpful if you would tell us which host distro you're using. Anyway, remove libfdt and related packages from your system. I had this kind of problems on Arch in the past.

mo123, 2021-06-16:
Ubuntu 20.04
I removed libfdt-dev package and now the build process continues.
Must have been an Ubuntu update because yesterday everything was still working fine.

Thanks

closed 2021-06-16 by jernejsk