LibreELEC/LibreELEC.tv #5450: restore backup doesnt work

s0i37, 2021-06-27:
### Describe the bug
restore backup doesnt work

### To Reproduce
Steps to reproduce the behavior:
1. Go to 'create backup'
2. Play 'restore backup'
3. ~/.kodi/userdata/addon_data/ are empty after restore

### Informations
 - LE Version: LibreELEC (official): 9.2.6 (RPi4.arm)
 - Hardware Platform: from RPi3B+ to RPi3

### Log file

### Additional context

antonlacon, 2021-06-28:
Does `~/.kodi/userdata/addon_data/` have contents in the backup tar file?

heitbaum, 2021-08-04:
@s0i37 - just following up on this ticket. How did you go?

heitbaum, 2021-08-05:
Closing. Please let us know any further issue and we can reopen if needed.

closed 2021-08-05 by heitbaum

DoctorOctagonapus, 2021-09-04:
If you're reading this and you're trying to restore a backup to downgrade from 10.0.0 to 9.x, don't bother. It won't work and you'll end up bricking your system. Copy the backup file off to an external disk, wipe and reinstall with the old version and restore the backup to that.