LibreELEC/LibreELEC.tv #5454: libreelec 9.95 and earlier plugins don't work until system is rebooted

swwdivernz, 2021-06-29:
Hi,
I have an interesting issue in that Kodi / LibreElec 9.9x will not always run the plugins and when starting one, it just gets the rotating  wheel when doing a power on and waiting for all services to start.

If I reboot it, the plugins will work OK. The first power on issue doesn't always do this but most times it will.  The next reboot by the shutdown menu will always work.

I am running it on X86 hardware and is a Asus EEEBox EB1501 unit that has HDMI. I think it has an Nvidia chipset. Once it is up an running after the reboot, it works fine.
It's like a service or some other hardware dependency is not running at first boot.

Set up logging but nothing of great significance showing. 

Affects all plugins.

This system used to work fine on earlier versions of Kodi / LibreElec (Leia) was fine. Went to 19 as Netflix etc needed Python 3 as was getting popups about dependencies not being met and Netflix had bad voice sync issues. Fixed that by going to raw mode.

Seems like this Kodi 19 has lots of problems.

Any Ideas?
Thanks in advance.

Cheers,

Simon

heitbaum, 2022-02-17:
Known issue with the paths, no fix presently.

heitbaum, 2023-09-06:
Some improving (libraries) expected with #7602 - but path won’t be addressed.

heitbaum, 2023-12-22:
I’m going to close this issue, now that the libraries are now self contained. For the bin directory there are no direct fixes - without a reboot/restart of Kodi.

closed 2023-12-22 by heitbaum