LibreELEC/LibreELEC.tv #5472: [BUG]

angrynerd, 2021-07-04:
### Describe the bug
Wireguard via connman does not work with ddns domain.

### To Reproduce
Steps to reproduce the behavior:
1. set the host as a domain.
2. does not connect.
3. set host to the public ip. 
4. does connect.

### Informations
 - LE Version: 9.2.6
 - Hardware Platform: RPI3b+

### Additional context
DDNS is very important personally as my ip address is not static with my isp.

lrusak, 2021-07-04:
Can you resolve the ddns domain with dig or curl or similar?

mglae, 2021-07-05:
There is some discussion on the forum:
https://forum.libreelec.tv/thread/17965-add-wireguard-support/
https://forum.libreelec.tv/thread/21603-wireguard-server-as-hostname-issue/
https://forum.libreelec.tv/thread/22007-wireguard-troubleshooting-no-reply/

Seem to be a connman limitation.

angrynerd, 2021-07-05:
@lrusak Yes I am amble to resolve the domain. @mglae I have read through the first one and tried to use the scripts from there to solve the issue but I dont understand them completely so im not sure how to fit them to my needs. The script in the second link seems like while its more files involved it should be simpler to understand. Im going to try that and then report back.

angrynerd, 2021-07-05:
Ok can confirm that the script from the second link works flawlessly! THank you for you assistance. To anyone else who hits this wall that script is largely the easiest to understand.

closed 2021-07-05 by angrynerd

g6094199, 2026-11-04:
as is stated on tasmota wiregurad support topic: https://github.com/arendst/Tasmota/issues/10022

"fyi

wireguard NEEDS to go with a network watchdog if the endpoint is a ddns and it resides on a dynamic ip. wireguard only resolves the name once. it doesnt track ip changes. so you need to track the endpoint via a watchdog and restart the wiregurad service or simply the device on a endpoints ip change.

**this info also should go to the wiki**."

openwrt used a script for this geting called eg every minute