LibreELEC/LibreELEC.tv #5486: [BUG] llvm target build ignores C_FLAGS AND CXX_FLAGS

heitbaum, 2021-07-09:
llvm cmake target options are being ignored.

Reference - https://gitlab.kitware.com/cmake/cmake/-/blob/master/Help/release/3.20.rst#L297-299

the following code is silently ignored in cmake < 3.20

https://github.com/LibreELEC/LibreELEC.tv/blob/b15ea798953a81ac7233f2455090e8600e0275b8/packages/lang/llvm/package.mk#L52-L53

it causes a fatal error in cmake 3.20 >=

what is the correct solution for the llvm/package.mk file? Removing the 2 lines would match existing behaviour - but assume that was not the intent.

can be tested with cmake 3.20.4
```
diff --git a/packages/devel/cmake/package.mk b/packages/devel/cmake/package.mk
index d0986997f9..3767ba40f2 100644
--- a/packages/devel/cmake/package.mk
+++ b/packages/devel/cmake/package.mk
@@ -3,8 +3,8 @@
 # Copyright (C) 2016-present Team LibreELEC (https://libreelec.tv)
 
 PKG_NAME="cmake"
-PKG_VERSION="3.19.2"
-PKG_SHA256="e3e0fd3b23b7fb13e1a856581078e0776ffa2df4e9d3164039c36d3315e0c7f0"
+PKG_VERSION="3.20.4"
+PKG_SHA256="87a4060298f2c6bb09d479de1400bc78195a5b55a65622a7dceeb3d1090a1b16"
 PKG_LICENSE="BSD"
 PKG_SITE="https://cmake.org/"
 PKG_URL="https://cmake.org/files/v$(get_pkg_version_maj_min)/cmake-${PKG_VERSION}.tar.gz"
```

mglae, 2021-07-09:
> Removing the 2 lines would match existing behaviour - but assume that was not the intent.

Reading https://cmake.org/cmake/help/latest/variable/CMAKE_LANG_FLAGS.html the intention is likely the default cmake behavior.

vpeter4, 2021-07-10:
The problem here is passing `PKG_CMAKE_OPTS_TARGET` as string to `cmake` which doesn't evaluate it's value correctly. It is general buildsystem problem and not llvm related. Something to consider for other packages too.

One possible solution is to use eval but it is considered dangerous.
```
diff --git a/scripts/build b/scripts/build
index 074cd8544c..60d30611ae 100755
--- a/scripts/build
+++ b/scripts/build
@@ -280,7 +280,7 @@ else
     # cmake builds with ninja
     "cmake:target")
       echo "Executing (target): cmake ${CMAKE_GENERATOR_NINJA} ${TARGET_CMAKE_OPTS} ${PKG_CMAKE_OPTS_TARGET} $(dirname ${PKG_CMAKE_SCRIPT})" | tr -s " "
-      cmake ${CMAKE_GENERATOR_NINJA} ${TARGET_CMAKE_OPTS} ${PKG_CMAKE_OPTS_TARGET} $(dirname ${PKG_CMAKE_SCRIPT})
+      eval cmake ${CMAKE_GENERATOR_NINJA} ${TARGET_CMAKE_OPTS} ${PKG_CMAKE_OPTS_TARGET} $(dirname ${PKG_CMAKE_SCRIPT})
       ;;
     "cmake:host")
       echo "Executing (host): cmake ${CMAKE_GENERATOR_NINJA} ${HOST_CMAKE_OPTS} ${PKG_CMAKE_OPTS_HOST} $(dirname ${PKG_CMAKE_SCRIPT})" | tr -s " "
```
The other is to use PKG_CMAKE_OPTS_TARGET as array which is prefered way of dealing _bash quoting hell_ problems.

```
diff --git a/packages/lang/llvm/package.mk b/packages/lang/llvm/package.mk
index 28e88dd0a0..d9d2f815ec 100644
--- a/packages/lang/llvm/package.mk
+++ b/packages/lang/llvm/package.mk
@@ -48,11 +48,12 @@ pre_configure_host() {
 }
 
 pre_configure_target() {
-  PKG_CMAKE_OPTS_TARGET="${PKG_CMAKE_OPTS_COMMON} \
+  PKG_CMAKE_OPTS_TARGET=(${PKG_CMAKE_OPTS_COMMON} \
                          -DCMAKE_C_FLAGS="${CFLAGS}" \
                          -DCMAKE_CXX_FLAGS="${CXXFLAGS}" \
                          -DLLVM_TARGET_ARCH="${TARGET_ARCH}" \
-                         -DLLVM_TABLEGEN=${TOOLCHAIN}/bin/llvm-tblgen"
+                         -DLLVM_TABLEGEN=${TOOLCHAIN}/bin/llvm-tblgen)
+
 }
 
 make_host() {
```

But then llvm:target doesn't build anymore.

mglae, 2021-07-11:
@vpeter4 Good catch.

> But then llvm:target doesn't build anymore.

`cmake ... ${PKG_CMAKE_OPTS_TARGET[@]} ...` is required to get the complete array. Fortunately this is "backward  compatible" and can expand a normal variable.

mglae, 2021-07-11:
Conclusion: to fix the llvm error and get the wanted parameters remove the -DCMAKE_C_FLAGS= and -DCMAKE_CXX_FLAGS= lines.

vpeter4, 2021-07-11:
One option is to use configure_target() function for llvm. This way PKG_CMAKE_OPTS_TARGET can be array.

vpeter4, 2021-07-12:
Like this

```
diff --git a/packages/lang/llvm/package.mk b/packages/lang/llvm/package.mk
index 28e88dd0a0..3320b0d8c8 100644
--- a/packages/lang/llvm/package.mk
+++ b/packages/lang/llvm/package.mk
@@ -34,6 +34,7 @@ PKG_CMAKE_OPTS_COMMON="-DCMAKE_BUILD_TYPE=MinSizeRel \
                        -DLLVM_ENABLE_ASSERTIONS=OFF \
                        -DLLVM_ENABLE_WERROR=OFF \
                        -DLLVM_ENABLE_ZLIB=ON \
+                       -DLLVM_ENABLE_LIBXML2=OFF \
                        -DLLVM_BUILD_LLVM_DYLIB=ON \
                        -DLLVM_LINK_LLVM_DYLIB=ON \
                        -DLLVM_OPTIMIZED_TABLEGEN=ON \
@@ -48,11 +49,16 @@ pre_configure_host() {
 }
 
 pre_configure_target() {
-  PKG_CMAKE_OPTS_TARGET="${PKG_CMAKE_OPTS_COMMON} \
+  PKG_CMAKE_OPTS_TARGET=(${PKG_CMAKE_OPTS_COMMON} \
                          -DCMAKE_C_FLAGS="${CFLAGS}" \
                          -DCMAKE_CXX_FLAGS="${CXXFLAGS}" \
                          -DLLVM_TARGET_ARCH="${TARGET_ARCH}" \
-                         -DLLVM_TABLEGEN=${TOOLCHAIN}/bin/llvm-tblgen"
+                         -DLLVM_TABLEGEN=${TOOLCHAIN}/bin/llvm-tblgen)
+}
+
+configure_target() {
+  echo "Executing (target): cmake ${CMAKE_GENERATOR_NINJA} ${TARGET_CMAKE_OPTS} "${PKG_CMAKE_OPTS_TARGET[@]}" $(dirname ${PKG_CMAKE_SCRIPT})" | tr -s " "
+  cmake ${CMAKE_GENERATOR_NINJA} ${TARGET_CMAKE_OPTS} "${PKG_CMAKE_OPTS_TARGET[@]}" $(dirname ${PKG_CMAKE_SCRIPT})
 }
 
 make_host() {
```

heitbaum, 2021-07-13:
@vpeter4 - checked it out - looks good in my dev tree using your code 4400a9282756f36c988eae447d34900885fe4db0

http://paste.ubuntu.com/p/2PMcD9B4nc/ Line 3654
`
Executing (target): cmake -GNinja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_TOOLCHAIN_FILE=/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/etc/cmake-x86_64-libreelec-linux-gnu.conf -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=MinSizeRel -DCMAKE_BUILD_TYPE=MinSizeRel -DLLVM_INCLUDE_TOOLS=ON -DLLVM_BUILD_TOOLS=OFF -DLLVM_BUILD_UTILS=OFF -DLLVM_BUILD_EXAMPLES=OFF -DLLVM_INCLUDE_EXAMPLES=OFF -DLLVM_BUILD_TESTS=OFF -DLLVM_INCLUDE_TESTS=OFF -DLLVM_INCLUDE_GO_TESTS=OFF -DLLVM_BUILD_BENCHMARKS=OFF -DLLVM_BUILD_DOCS=OFF -DLLVM_INCLUDE_DOCS=OFF -DLLVM_ENABLE_DOXYGEN=OFF -DLLVM_ENABLE_SPHINX=OFF -DLLVM_ENABLE_OCAMLDOC=OFF -DLLVM_ENABLE_BINDINGS=OFF -DLLVM_TARGETS_TO_BUILD=AMDGPU -DLLVM_ENABLE_TERMINFO=OFF -DLLVM_ENABLE_ASSERTIONS=OFF -DLLVM_ENABLE_WERROR=OFF -DLLVM_ENABLE_ZLIB=ON -DLLVM_ENABLE_LIBXML2=OFF -DLLVM_BUILD_LLVM_DYLIB=ON -DLLVM_LINK_LLVM_DYLIB=ON -DLLVM_OPTIMIZED_TABLEGEN=ON -DLLVM_APPEND_VC_REV=OFF -DLLVM_ENABLE_RTTI=ON -DLLVM_ENABLE_UNWIND_TABLES=OFF -DLLVM_ENABLE_Z3_SOLVER=OFF -DCMAKE_C_FLAGS=-march=x86-64 -m64 -mmmx -msse -msse2 -mfpmath=sse -Wall -pipe -O2 -fomit-frame-pointer -DCMAKE_CXX_FLAGS=-march=x86-64 -m64 -mmmx -msse -msse2 -mfpmath=sse -Wall -pipe -O2 -fomit-frame-pointer -DLLVM_TARGET_ARCH=x86_64 -DLLVM_TABLEGEN=/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/bin/llvm-tblgen /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/build/llvm-12.0.1
`