LibreELEC/LibreELEC.tv #5487: Constant load of ~35% of 1 CPU core on Rock64

diederikdehaas, 2021-07-09:
### Describe the bug
I noticed that the load on 1 CPU core on a Rock64 is constantly around 35% while AFAICT it's not doing anything.
The Rock64 (already) has a heat problem and this is ofc not helping. While idling, the temp is usually around 64C.
With the help of 'ndufresne' on #libreelec IRC channel I gathered some data which may help.
I removed the '[BUG]' prefix of this issue as I'm not sure it's an actual bug.

### To Reproduce
Steps to reproduce the behavior:
1. Go to Settings -> System Information -> Hardware
2. Watch the CPU load and temperature
3. See load is constantly ~35% on 1 CPU core (the busy core does change from time to time) and see that the the temp is ~64C.

### Informations
 - LE Version: nightly-20210705-7e5854c (RK3328.arm)
 - Hardware Platform: Pine64's Rock64
 - kernel: Linux kodi-rock64 5.10.47 #&#x2060;1 SMP Mon Jul 5 19:05:21 CEST 2021 aarch64 GNU/Linux

### Log file
I've attached a log file (libreelec-htop-perf-output.txt) with the output of `top` which shows `kodi` is causing the load and the output of `perf top -g`. On my (main) PC `perf` shows an output similar to `top`, but on my Rock64 device instead of updating the same line(s), it added them add the end. The full output is pasted at https://pastebin.com/yiAjamjL (but only available for 13 more days), but in the attached file I removed a lot of whitespace to make it smaller and hopefully more useful.

The LE-perf-p-output.txt log file is the output when I ran `perf top -g -p <PID-of-kodi>`

[libreelec-htop-perf-output.txt](https://github.com/LibreELEC/LibreELEC.tv/files/6793547/libreelec-htop-perf-output.txt)
[LE-perf-p-output.txt](https://github.com/LibreELEC/LibreELEC.tv/files/6793667/LE-perf-p-output.txt)

diederikdehaas, 2021-07-09:
On IRC I also asked whether there is a `toram` option, but got no response to that.
That was in relation to these comments by 'ndufresne':
```
squashfs overhead really ?
it's doing massive amount of read()
```
As I have the 4GB variant of the Rock64, there's a lot of RAM that seems unused.
```
kodi-rock64:~ # free -h
               total        used        free      shared  buff/cache   available
Mem:           3.9Gi       226Mi       3.3Gi       157Mi       408Mi       3.5Gi
Swap:             0B          0B          0B
```
I don't know if it could be relevant, but figured I'd mention it anyway.

lrusak, 2021-07-09:
Not much we can do. Kodi is a heavy multi threaded application (20+ threads) that's is constantly running background tasks (timers, add-on updates, input monitoring, etc).

The best bet may be to either play with dirty regions (advancedsettings.xml) or see if you can stop other tasks (RSS, service addons, etc).

You can also use `top -H <PID>` to check if a specific thread is overactive.

closed 2021-07-09 by lrusak

diederikdehaas, 2021-07-09:
I was able to use `visualizedirtyregions` parameter, but if I use `algorithmdirtyregions`, iow literally https://kodi.wiki/view/HOW-TO:Modify_dirty_regions#Example_1, then the screen gets stuck at the loading screen with LibreELEC logo/name and version in the top-right corner.

knaerzche, 2021-07-10:
Note that Rockchip SoCs using ondemand scaling govenor for cpu cores in LE - for RK3328 that means, cpu might be running as low as 408 MHz when idle - so 35% usage of one core is really not much in that case.
Since Rock64 board comes without: You might want install a heatsink (64°C is far from being critical, btw.)