LibreELEC/LibreELEC.tv #5496: 4K runs slowmotion and only 1920 x 1080

petcol, 2021-07-18:
### Describe the bug
Bauhn 4K 75" panel will not operate using Raspberry Pi 4 B at 4K

### To Reproduce
Steps to reproduce the behavior:
Run up the image file onto MicroSD card Version used is LibreELEC-RPi4.arm-9.2.7 
Only displays at 1920  x 1080 playing any 4K content local Disc or network is like in slow motion

### Informations
 - LE Version: 9.2.7
 - Raspberry Pi 4 B - 8 RAM

I wasn't able to fins the config.txt file in the boot partition?

### Informations 2
LibreELEC-RPi4.arm-9.95.5.img  [ beta ]
 - Raspberry Pi 4 B - 8 RAM

I was able to find the config.txt file in the boot partition and I added the hdmi_enable_4kp60=1 to the end of the config save and once booted the panel start showing it was at the higher resultion, once Kodi loaded and and went in to setting to check the display resolution it was also showing the 4K setting I'm not sure exactly what I did, I think I changed the 2 Ch audio to 7.1 to match my receiver and after that the panel was displaying 1920 x 1080 and will not change.

I'm about to re-image the card and try to reproduce the problem

chewitt, 2021-07-19:
Please retest with LE10 images and use the forums for support/follow-up, thanks.

petcol, 2021-07-19:
https://forum.libreelec.tv/thread/24255-4k-runs-slow-motion-and-only-1920-x-1080/

closed 2021-07-19 by chewitt