LibreELEC/LibreELEC.tv #5500: [BUG] Outdated NOOBS packages

isantolin, 2021-07-19:
### Describe the bug
The latest version for LibreELEC is 9.2.8 but Raspberry NOOBS downloads 9.2.6
Anyone can update Packages for Noobs? http://releases.libreelec.tv/noobs/LibreELEC_RPi2/System.tar.xz, http://releases.libreelec.tv/noobs/LibreELEC_RPi2/Storage.tar.xz

### To Reproduce
1. Go to Raspberry NOOBS LibreELEC Install
2. Installed version will be 9.2.6, not 9.2.8

### Informations
 - LE Version: Latest NOOBS
 - Hardware Platform: RPi3

CvH, 2021-07-19:
this is intended, at the moment at least
9.2.7 and 9.2.8 just include experimental fixes targeted for the usage of widevine, nothing else
in the meantime you can just update to the latest version if needed

chewitt, 2021-09-14:
Closing as RPi/RPi2 are now running 9.2.8 and RPi4 is updated to 10.0.0

closed 2021-09-14 by chewitt