LibreELEC/LibreELEC.tv #5508: Since default 4kp60=1 on RPi4B no boot screen and only 4K 45Hz

dkadioglu, 2021-07-24:
I follow the nightly builds on the RPi4B and was curious, when I read that 4K 60Hz would be enabled finally. Unfortunately since that happened, I cannot see any boot screen anymore, i.e. no command line output and no libreelec logo. This means, that I also don't see the console output during an update of the image. Furthermore, I only can choose 3840x2160 with max 45 Hz in KODI, 60 Hz is not offered.

My Setup:
- RPi4B 8GB
- Panasonic TV TX55-FXW784
- Samsung HW-MS650 Soundbar
- RPi HDMI0 (next to USB-C) is connected to HDMI in of Soundbar and HDMI out of Soundbar is connected to TV HDMI 2
- TV and Soundbar should support 4K 60Hz as far as I understand

Output of modetest: http://ix.io/3tXm

If I can supply anything else helpful, please ask.

HiassofT, 2021-07-24:
The specs of your soundbar state that 4kp50/60 is only supported with 4:2:0 chroma subsampling, RPi4 requires 4:4:4.

Your TV seems to support 4:4:4, usually that needs to be enabled in the TV's settings menu though ("ultra deep colour" or some similar setting) and might not be supported on all HDMI ports.

Try connecting the RPi4 directly to the TV with a HDMI cable specced for 18gbit/sec bandwidth then 4kp50/60 should work.

Not sure where the 4k 45Hz mode is coming from, this is really weird, I suspect your soundbar might be adding that to the EDID.

dkadioglu, 2021-07-24:
Yes, you're right. After connecting directly to the TV, 4K@60Hz worked. However, the weird thing is, that I already tried that, without success. Don't know, what's different now. So what, works now. But still, there is the problem, that with the connection via the Soundbar and 4kp60=1 there is no console output and boot screen logo. As I said, without this, you are not able to see console output during an update, so you don't know, if it runs successfully until the hopefully successful reboot. Could this behaviour be fixed somehow?

HiassofT, 2021-07-25:
Can you post modetest output when connected directly to the TV? I suspect it won't contain the 45Hz mode (and kodi also doesn't list the 45Hz mode).

My guess is the soundbar is adding that bogus 45Hz mode (probably a firmware bug), without 4kp60 enabled that mode got filtered out (as the pixel clock is too high) but with 4kp60 enabled it gets chosen as it's flagged as being the preferred mode in EDID.

If that's the case you could check if there's some firmware update available for your soundbar

dkadioglu, 2021-07-30:
To give you an update: I ordered new cabling and now the RPi is directly connected to TV and 4K@60Hz works as it should. The Soundbar is connected to the TV and Sound is working via ARC for now. And you were right, via the direct connection between TV and RPi, the bogus 45Hz mode is not contained in the modetest output anymore.
But now, I have the same problem as described in https://github.com/raspberrypi/linux/issues/4486. I hope this can be fixed. If you need further info regarding this or want me to test something, please ask.

HiassofT, 2021-07-30:
Glad you got it working now!

I'd recommend configuring the (GUI) resolution to 1920x1080 and set up the whitelist (so kodi can switch to 4k etc when needed). Then you won't hit the issue when you are in the GUI and if you put your TV into standby when playing a 4kp50/60 video you can easily get screen back by pressing stop (kodi will then switch to 1920x1080).

FYI: we just stopped enabling 4kp60 by default because of that issue - change is already in latest nightly builds.

dkadioglu, 2021-08-25:
According to https://github.com/raspberrypi/linux/pull/4538 this should be fixed now. Could this go into LE 10 final, before it is released?

HiassofT, 2021-08-25:
The fix came too late for 10.0.0, it will be included in 10.0.1

dkadioglu, 2021-08-28:
> The fix came too late for 10.0.0, it will be included in 10.0.1

Ok, thanks for that info! I just built it myself and it seems to work now, so no more black screen after switching off and on my tv. Furthermore, 4k@60hz to hdmi0 and audio to hdmi1 works, too.
The only open issue is still, that with 4kp60 I don't see the splash screen and boot messages (like shell output during update) anymore, so the first screen output during boot is the kodi Matrix splash screen. Any idea, what could be the reason for this and how this could be fixed?

HiassofT, 2021-08-28:
Thanks a lot for testing and reporting back, it's good to know the black screen issue is fixed on your setup, too!

I ran into the "no splash / boot messages" issue as well, it happens when hdmi1 is connected. See this RPi kernel issue https://github.com/raspberrypi/linux/issues/4540

dkadioglu, 2021-10-17:
I am using builds which include the fix mentioned in raspberrypi/linux#4540 since a couple of weeks now and the issue of having no splash screen is fixed for me. I'll close this now. Thank you very much for looking into this!

closed 2021-10-17 by dkadioglu