LibreELEC/LibreELEC.tv #5509: [BUG] LE 9.2.{6,7,8} won't boot at all on Pi 4 v1.4

coaxial, 2021-07-24:
### Describe the bug
LibreElec 9.2.{6,7,8} won't boot at all on a Pi 4 rev 1.4

### To Reproduce
Steps to reproduce the behavior:
1. Use Balena Etcher to flash https://releases.libreelec.tv/LibreELEC-RPi4.arm-9.2.8.img.gz (via the URL option)
2. Insert SD card into a Pi 4 v1.4 (running any bootloader version)
3. Pi starts up and gets stuck to a black screen with a stuck, white `_` in the top left corner

### Informations
 - LE Version: 9.2.6 to 9.2.8
 - Hardware Platform: Pi 4 v1.4

### Log file

Unable to get any log files, the Pi just freezes before LE can do anything.

### Additional context

This has been reported by several users on the forum with a wide combination of SD cards, PSUs (including the official PSU), and Pi 4 v1.4 boards.

Is the firmware LE 9.2 ships outdated? I tried swapping `start.elf` and `fixup.dat` from the SD card with the files in the `boot/` folder from here: https://codeload.github.com/raspberrypi/firmware/zip/refs/heads/stable but no luck, the Pi blinks 4 long 4 short.

HiassofT, 2021-07-24:
Please use LE10. We won't add support for newer devices to LE9.2

closed 2021-07-24 by HiassofT

coaxial, 2021-07-24:
I will, once it's not in beta anymore. In the meantime, I don't know what else people in that situation can do.

chewitt, 2021-07-24:
The release notes for LE 10.0.0 will include a footnote for RPi stating we still consider things to be experimental in places so if you're going to get hung-up on release numbers and labels you'll have a long wait, LE11 perhaps. Or you could just use it now, since it's perfectly usable for 95% of use-cases.