LibreELEC/LibreELEC.tv #5512: [BUG]

jclemons7, 2021-07-29:
RPi2 Won't boot past rainbow screen with LibreELEC-RPi2.arm-9.2.8.img.gz 

### To Reproduce
Load LibreELEC-RPi2.arm-9.2.8.img.gz on SD card.  Boot.

### Informations
 - 9.2.8
 - RPi2
 - 8GB SD card

### Log file
Can't get log file, won't boot

### Additional context
RPi2 boots fine with other OS's with exact same physical configuration.

lrusak, 2021-07-29:
How did you flash the SD card?

jclemons7, 2021-07-29:
I used LibreELEC.USB-SD.Creator.Linux-64bit.bin, but now I'm trying Imager, which seems to have an older version of Elec.. more to come on that I guess.

jclemons7, 2021-07-29:
Hmmm.. same effect with Imager.

antonbo, 2022-01-03:
Hi,
I have same issue. But in my case the problem appears with RPi2 B+ v1.2. Older version v1.1 (which uses another chip) works fine with same SD card and power supply. Seems like a problem is in Linux kernel. I've tried to put kernel.img from Raspbian Bullseye and it started.

heitbaum, 2022-02-17:
Closing as a stale ticket. Please reopen with updated info, or use forum.

closed 2022-02-17 by heitbaum