LibreELEC/LibreELEC.tv #5521: [BUG] Upgrading from 9.95.5 to 9.97.1 breaks HECV streaming in Moonlight-qt

veldenb, 2021-07-31:
### Describe the bug
Upgrading my Pi 4 from 9.95.5 to 9.97.1 breaks the HEVC decoding in Moonlight-qt, vertical tiles appear. In 9.95.5 it works fine.
It looks like this: [image] 

### To Reproduce
Steps to reproduce the behavior:
1. Install LibreELEC 9.95.5 on Pi4
2. Install this Moonlight-qt plugin: https://github.com/veldenb/plugin.program.moonlight-qt
3. Start Streaming, HEVC decodes normal
4. upgrade LibreELEC to 9.97.1 on Pi4
5. The stream looks tiled

### Informations
 - LE Version: 9.97.1
 - Hardware Platform: PRi4

### Log file
[moonlight-qt.log.kodi.9.95.5.txt](https://github.com/LibreELEC/LibreELEC.tv/files/6911165/moonlight-qt.log.kodi.9.95.5.txt)
[moonlight-qt.log.kodi.9.97.1.txt](https://github.com/LibreELEC/LibreELEC.tv/files/6911164/moonlight-qt.log.kodi.9.97.1.txt)
Kodi is stopped while running moonlight so no logs there.

### Additional context
See more information abouw the bug here: https://github.com/veldenb/plugin.program.moonlight-qt/issues/9
It looks like things went wrong somewhere here: https://github.com/LibreELEC/LibreELEC.tv/compare/9.95.5...9.97.1

HiassofT, 2021-07-31:
Please report this to the moonlight-qt plugin maintainers. Most likely it's embedded ffmpeg needs to be updated to work with latest RPi kernel.

closed 2021-07-31 by HiassofT

veldenb, 2021-07-31:
Thanks for your reply, I'll do that.