LibreELEC/LibreELEC.tv #5535: [BUG] Cross Compiled Python3 stores the CFLAGS causing failure of the cross-compiler if the CFLAGS are not supported.

heitbaum, 2021-08-07:
### Describe the bug

When compiling for riscv the following Python3 addons fail:
- pycryptodome
- Pillow3

From my reading the **CFLAGS** are compiled into the Python3 binary and are subsequently used to compile modules: `CFLAGS = "-Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=native -O2 -Wall -pipe -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-D1.riscv64-10.0-devel/toolchain/include -Wno-format-security -march=native -O2 -Wall -pipe -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-D1.riscv64-10.0-devel/toolchain/include -Wno-format-security"`

- See full `python -m sysconfig` here http://paste.ubuntu.com/p/CY9TNxHSzQ/

In this case `-march=native` is not supported by the `riscv64-libreelec-linux-gnu-gcc` compiler.

The output of the `PROJECT=RiscV DEVICE=D1 ARCH=riscv64 UBOOT_SYSTEM=nezha s/build Pillow` is here: http://paste.ubuntu.com/p/yn5t3Hvb4H/

### Information
 - LE Version: LE10+
 - Hardware Platform: x86_64 TGL buildhost / rv64gc target

fuzzard, 2021-08-07:
Have fun with this. The following is what I ended up doing for Kodi with the exact issue.

https://github.com/xbmc/xbmc/pull/19246/

It's a long dark rabbit hole when you dig into this shit. Ben Fogles Crossenv ended up being by far the easiest and cleanest way I could handle python's build system.
Essentially it allows to even just use pip to build target modules if desired.

Caveat on all this, I still have unresolved issues with android builds and PIL that are the reason this hasn't gone into Kodi core yet, but I believe this is packaging issues more so than build issues, as all kodis apple platforms work a treat with this

heitbaum, 2021-08-07:
I did think it was going to go that way ☹️ Thanks for the couple of tips. I did briefly read Ben Fogles Crossenv earlier this week, but just put some notes aside until I had the ARCH compiling through to the end. Now time to document the couple of hacks I had done to get it through.