LibreELEC/LibreELEC.tv #5542: [Idea] Integrate docker/... into LE build system

CvH, 2021-08-09:
We are facing build problems since day one due the current approach of the build system. 
The build system is heavily host dependent and we are regular facing not reproduceable build problems. 
There is no proper solution to it because every package could fetch something from the host system - or not. At every update it could start to do it - or not. Everyone uses a different ENV to build LE, so the result always differs.

So the idea came up again to integrate docker (or some other chroot) into the LE build system so that the normal build runs at a docker container instead "at" the host system. 

Main benefit is reproducibility, so everyone has the same build environment, everyone facing the same problems.

I am aware that you can manually wrap docker/LXC/whatnot around the LE build ENV.

From my point of view the main goal should be that everyone uses the same build env to build LE - out of the box.

I am looking for opinions, as far as I know nobody has put time into that yet, but if the general idea is acceptable someone will maybe look into that.

vpeter4, 2021-08-09:
So what exactly you are saying? To make some small build environment which would actually build docker and then LE inside of it? Without knowing anything about docker?
Something like that is used at company I work for. Mainly because of easier set up of a build system.

heitbaum, 2021-08-10:
I have been doing builds in docker since the start. I had started that way ... 

I have 2 x NUC6 - one VMware ESXi and one was my LE9, use to have a docker to have a full os on the LE9. When I started doing the development leading up to LE10 and that the NUC6 was performing badly on some media playback and we ended up with the 4K TV ---> The NUC11 16GB and now 64GB TGL arrived. So the TGL our daily LE10+ device and it has the media library connected to it via USB. I have the N2+, the NUC6 and one of the other "SBC devices" running on the TV now. So if there is a something to break - can always go back and forth to a device to watch media. Then on my desk - 1 or 2 devices is some state of destruction :-)

In my daily LE work. I ssh to NUC11 and run `./scripts/docker.sh` which takes me to the repo `docker@83d6839460d3:/var/media/DATA/home-rudi/LibreELEC.tv/` and I package/develop away. As the kernel does get exposed into the container - I had the 5.12 build issue early on - once I have updated my LE10 host to 5.12 kernel (TGL issues ...)

I have 3 other directories with their own branches
```
LibreELEC.binaries - used to run "tools/mkpkg/update_binary-addons"
LibreELEC.dev - my main dev tree
LibreELEC.kernel10rc - track the kernel
LibreELEC.tv - build tree for the required target of the moment. --- git co ABC ---
```

**Example of building the librespot when we still had the rust issue:**
```
$ more 10-build.sh
docker build --pull -t groovy tools/docker/groovy
docker run --rm -v `pwd`/..:/build -w /build/LibreELEC.tv -it \
 -e PROJECT=Generic -e ARCH=x86_64 -e ADDON_JENKINS=yes groovy scripts/create_addon service.librespot
```

I'm supportive of the docker because of the above and the couple anomalies that were identified with libraries / ... (e.g. stability of chrome build) on build2. I do think a stable minimal os with the bare needs to create the toolchain is ideal. But with a fully featured FAT os to allow development on the same lines for day-to-day development. so my `./scripts/docker.sh` is a FAT version of "Ubuntu 18.04.5 LTS" - that I have added to from the `tools/docker/bionic/Dockerfile` base. If I do a sudo apt install ABC and want to keep it - I run `docker commit XXXX le-dev:XX` which is then launched next time I run `./scripts/docker.sh`. I have many of the `le-dev:latest` containers running at any one time - but know that they are all "stateless" - but accessing the same /data path.

Anyhow - my 2c and how I have been working.

CvH, 2021-08-10:
> What base OS would be preferred for the docker container?

Likely something Debian based (aka standard) like Debian/Ubuntu ... 

> So what exactly you are saying?

You would hit `PROJECT=Generic ARCH=x86_64 make image` and the LE build system would pull (for example) a Ubuntu docker and runs the build commands inside of the docker while all build files are mapped to the "outside".
So basically everything would stay where it is just the build itself is inside of a container.

vpeter4, 2021-08-10:
> So basically everything would stay where it is just the build itself is inside of a container.

As I though - some smart wrapper around build system without needed to understand the technology behind it.

nomandera, 2021-08-10:
This is all entirely possible. It would also be possible for the image to run the source download as part of the build so that when pulling a tagged image you would get both a guaranteed base OS and the source packages.

The more that can be tagged like this the better as it inherently makes for greater build commonality and easier bisecting

heitbaum, 2022-06-07:
Closing as completed:- https://github.com/LibreELEC/actions/actions

closed 2022-06-07 by heitbaum