LibreELEC/LibreELEC.tv #5550: cmdline.txt boot based on disk LABEL is too weak

efa, 2021-08-15:
### Describe the bug
the boot based on disk LABELS is too weak
changing the labels results in LE unbootable

### To Reproduce
Steps to reproduce the behavior:
1. installed LibreELEC
2. try if it can boot
3. shut it down
4. change the disk label of SYSTEM partition
5. try if it can boot: NO see log

### Informations
 - LE Version: v9.2.6
 - Hardware Platform: Raspberry Pi3B+

### Log file
the error on boot is:

*** Error in mount_flash: mount_common: Could not mount LABEL=System ***
### Starting debugging shell for boot step: mount_flash... type  exit  to quit ###
#

### Additional context
I tryed to set back the disk label partitions with:
$ sudo mlabel -v  -i /dev/mmcblk0p8 ::System
$ sudo e2label /dev/mmcblk0p9 Storage
but the error remian the same.
I made a try with:
$ sudo mlabel -v  -i /dev/mmcblk0p8 ::SYSTEM
just to be sure this is not related to short/long name on FAT32 disk label, but got same error.

Then I also tryed to set 'cmdline.txt' to:
boot=UUID=3E30-E976 disk=UUID=21a3db18-287e-4e80-8645-e0ec29971ffe debugging
where UUID are the one for System and Storage partitions, but then got the error log below:

Framebuffer vertical res: 1050
Framebuffer spalsh image: /splash/splash-1080.png
mount: mounting /flash/SYSTEM in /sysroot failed: No such file or directory
... (total 15 times)
mount: mounting /flash/SYSTEM in /sysroot failed: No such file or directory
*** Error in mount_sysroot: mount_common: Could not mount /flash/SYSTEM ***
### Starting debugging shell for boot step: mount_sysroot... type  exit  to quit ###
#

Should be somewhere somthing that continue to look for SYSTEM also with UUID set, and do not find the label also once restored.
The only thing I could do to solve the trouble was a reinstallation of LE.
I made a new try, but same issue, so it is reproducible.
As now I reinstalled LE and keep its partition labels as default, but it is confusing on an uSD with other OS.

I noticed on another installation of LE for Raspberry PI1B, the cmdline.txt was already based on UUID and infact I changed the disk labels many time without any trouble.

Note: I had other OSs also in other computers, and for all I changed the disk labels, and they work, labels are for humans
please solve this fragility

efa, 2021-08-15:
I looked at:
https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/sysutils/busybox/scripts/init
and then
NOOBS/os/LibreELEC_RPi2/partition_setup.sh
but I cannot understand where the parameters come from, sorry

lrusak, 2021-08-15:
The latest versions of LibreELEC all use UUID for disk mounting.

Of course changing the disk label in a label based mounting system would break the system, what did you expect?

closed 2021-08-15 by lrusak

efa, 2021-08-15:
I expect that setting back the partition labels to the original value OR setting the cmdline.txt to use UUID, the system boot

efa, 2021-08-15:
in May 2021 NOOBS 3.7 the file:
config/noobs/partition_setup.sh
set the cmdline.txt to use disk label, it is reproducible

lrusak, 2021-08-15:
Please use the forum for support. Thanks.

efa, 2021-08-19:
I made another try and changed the 'cmdline.txt' for use the UUIDs, BEFORE renaming the partition labels, and in this manner work.

I also asked to NOOBS guy to set $id1 and $id2 always with UUID before call 'partition_setup.sh', let's see.

In the mean time can you accept a PullReq that use 'blkid' or 'lsblk -o UUID' to always write the UUID also when labels are passed from NOOBS?

efa, 2021-09-11:
In the mean time can you accept a PullReq that use 'blkid' or 'lsblk -o UUID' to always write the UUID also when labels are passed from NOOBS?