LibreELEC/LibreELEC.tv #5577: [BUG] Ass subtitle not shown

rjousse18, 2021-08-28:
### Describe the bug
When i play video with .ass subtitle, I can't see them. Other subtitle type are display correctly

### To Reproduce
Steps to reproduce the behavior:
1. Play video with ass subtitle

### Informations
 - LE Version: 10.0.0
 - Hardware Platform: RPi4

### Log file
https://paste.kodi.tv/giqemidope.kodi

### Additional context

HiassofT, 2021-08-28:
This seems to be a kodi bug: https://github.com/xbmc/xbmc/issues/20085

I posted some info how to avoid or workaround the issue here on the forum: https://forum.libreelec.tv/thread/23951-libreelec-rpi4-9-95-2-does-not-reproduce-ass-subtitles/?postID=159495#post159495

rjousse18, 2021-08-29:
I'm on 4k TV, when I enable "adjust refresh rate" I have a very small display of the video (with subtitle !) :/ in your solution you said "disable GUI resolution limit", where can i find this option ?

chewitt, 2021-11-26:
Closing as this should have been auto-closed by #5703

closed 2021-11-26 by chewitt