LibreELEC/LibreELEC.tv #5588: connman fails to add static routes with dhcp + (workaround)

hvegh, 2021-08-29:
### Context
Due to the fact that my IPS provided modem is unable to configure an additional static route I had to resort to distribute static route to clients using dhcp.

On the DHCP server i added:
`dhcp-option= option:classless-static-route,0.0.0.0/0,192.168.1.253,192.168.100.0/24,192.168.1.1`

Tell the LibreElec client that 192.168.1.1 is able to figure out where to route 192.168.100.0/24.
Resulting route:

```
LibreELEC:~ # route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.1.253   0.0.0.0         UG    0      0        0 eth0
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 eth0
192.168.1.253   0.0.0.0         255.255.255.255 UH    0      0        0 eth0
```

Then I look at the connman status:
```
LibreELEC:~ # systemctl status connman.service
● connman.service - Connection service
   Loaded: loaded (/usr/lib/systemd/system/connman.service; disabled; vendor preset: disabled)
   Active: active (running) since Thu 2019-04-11 18:28:40 CEST; 2 years 4 months ago
 Main PID: 454 (connmand)
   Memory: 2.2M
   CGroup: /system.slice/connman.service
           └─454 /usr/sbin/connmand -nr --config=/etc/connman/main.conf

Apr 11 18:28:41 LibreELEC connmand[454]: eth0 {del} route fe80:: gw :: scope 0 <UNIVERSE>
Apr 11 18:28:41 LibreELEC connmand[454]: eth0 {del} route ff00:: gw :: scope 0 <UNIVERSE>
Apr 11 18:28:41 LibreELEC connmand[454]: Setting domainname to god.lan
Apr 11 18:28:41 LibreELEC connmand[454]: eth0 {add} address 192.168.1.85/24 label eth0 family 2
Apr 11 18:28:41 LibreELEC connmand[454]: ntp: adjust (jump): +75256701.268360 sec
Aug 29 19:07:02 LibreELEC connmand[454]: eth0 {add} route 192.168.1.0 gw 0.0.0.0 scope 253 <LINK>
Aug 29 19:07:02 LibreELEC connmand[454]: eth0 {add} route 192.168.1.253 gw 0.0.0.0 scope 253 <LINK>
Aug 29 19:07:02 LibreELEC connmand[454]: eth0 {add} route 0.0.0.0 gw 192.168.1.253 scope 0 <UNIVERSE>
Aug 29 19:07:02 LibreELEC connmand[454]: eth0 {add} route 82.165.8.211 gw 192.168.1.253 scope 0 <UNIVERSE>
Aug 29 19:07:04 LibreELEC connmand[454]: eth0 {del} route 82.165.8.211 gw 192.168.1.253 scope 0 <UNIVERSE>

```
So DHCP client of connman is unable to deal with dhcp option 121, unfortunately. This additional route 

### ISSUE

Would be nice to add dhcp option 121 to connman or maybe switch over to native systemd dhcp client?

Also it would be nice to add **EnableOnlineCheck=false** to  **/etc/connman/main.conf** in order to ged rid of the phone home of connman...

### Workaround

Create a service that starts after network is online in order to add the route.

```
# File: /storage/.config/system.d/fixup.service

[Unit]
Description=Local fixups
Before=kodi.service
Requires=network-online.service
After=network-online.service
ConditionPathExists=/storage/.config/fixup.sh

[Service]
Type=oneshot
Environment=HOME=/storage
ExecStart=-/bin/sh -c ". /etc/profile; exec /bin/sh /storage/.config/fixup.sh"
RemainAfterExit=No

[Install]
WantedBy=kodi.service
```
and

```
# File: /storage/.config/fixup.sh             
                                              
route add -net 192.168.100.0/24 gw 192.168.1.1

```
and finally:
`systemctl enable /storage/.config/system.d/fixup.service`

chewitt, 2021-08-29:
ping @igaw would it be possible to add dhcp option 121 support?

hvegh, 2021-08-30:
> ping @igaw would it be possible to add dhcp option 121 support?

See [rfc3442](https://datatracker.ietf.org/doc/html/rfc3442) for an explanation.
If needed, I can help creating an iterator function on the option field.

```
Classless Route Option Format

   The code for this option is 121, and its minimum length is 5 bytes.
   This option can contain one or more static routes, each of which
   consists of a destination descriptor and the IP address of the router
   that should be used to reach that destination.

    Code Len Destination 1    Router 1
   +-----+---+----+-----+----+----+----+----+----+
   | 121 | n | d1 | ... | dN | r1 | r2 | r3 | r4 |
   +-----+---+----+-----+----+----+----+----+----+

    Destination 2       Router 2
   +----+-----+----+----+----+----+----+
   | d1 | ... | dN | r1 | r2 | r3 | r4 |
   +----+-----+----+----+----+----+----+

   In the above example, two static routes are specified.

   Destination descriptors describe the IP subnet number and subnet mask
   of a particular destination using a compact encoding.  This encoding
   consists of one octet describing the width of the subnet mask,
   followed by all the significant octets of the subnet number.
   ...
   The number of significant octets is the
   width of the subnet mask divided by eight, rounding up, as shown in
   the following table:

        Width of subnet mask     Number of significant octets
                     0                     0
                  1- 8                     1
                  9-16                     2
                 17-24                     3
                 25-32                     4
```

igaw, 2021-08-31:
I'll check if this is reasonable simple to add. The parsing is not the problem. It's more the additional routing information needs
to be tracked somehow and hopefully we don't have to jump through several hoops to get this implemented.

heitbaum, 2025-02-27:
Did this feature request get implemented? Is it still valid/required?

chewitt, 2025-02-27:
I did some grep'ing of connman commit logs looking for obvious keywords and didn't see anything promising, so I don't think anything happened. The best thing for @hvegh to do is describe the issue to the ConnMan mailing list so some of the more active contributors see the report. It's a feature request more than a bug though.

igaw, 2025-02-27:
I don't think so, I've stopped contributing to the ConnMan project a while ago. Just didn't find the motivation to work on it in my spare time.

heitbaum, 2026-02-13:
Closing as known issue. To be solved upstream.

closed 2026-02-13 by heitbaum