LibreELEC/LibreELEC.tv #5603: [BUG] Segmentation fault when skipping inputstream.adaptive video

freynder, 2021-09-04:
### Describe the bug
Segmentation fault when skipping or resuming VOD (facilitated by inputstream.adaptive) on rpi4

### To Reproduce
1. Start VOD (in my case this is through VTM GO (Belgium Flemish TV) addon)
2. Skip a few minutes forward
3. Watch LE reboot due to crash

Same happens when resuming a VOD to the last recorded timestamp

### Informations
- Arch: aarch64
- Kernel: Linux 5.10.52 #1 SMP Wed Aug 11 14:10:51 CEST 2021
- Release: LibreELEC 10.0.0 
- Hardware Platform: RPI4

### Log file
crash logs are attached
They show:
```
Program terminated with signal SIGSEGV, Segmentation fault.
  57 #0  0xf69a9190 in ?? () from /usr/lib/libavcodec.so.58
```
[kodi_crashlog_20210904173319.log](https://github.com/LibreELEC/LibreELEC.tv/files/7110273/kodi_crashlog_20210904173319.log)
[kodi_crashlog_20210902231144.log](https://github.com/LibreELEC/LibreELEC.tv/files/7110264/kodi_crashlog_20210902231144.log)
[kodi_crashlog_20210903192433.log](https://github.com/LibreELEC/LibreELEC.tv/files/7110265/kodi_crashlog_20210903192433.log)
[kodi_crashlog_20210903193953.log](https://github.com/LibreELEC/LibreELEC.tv/files/7110266/kodi_crashlog_20210903193953.log)
[kodi_crashlog_20210903194035.log](https://github.com/LibreELEC/LibreELEC.tv/files/7110267/kodi_crashlog_20210903194035.log)
[kodi_crashlog_20210903194111.log](https://github.com/LibreELEC/LibreELEC.tv/files/7110268/kodi_crashlog_20210903194111.log)
[kodi_crashlog_20210903203500.log](https://github.com/LibreELEC/LibreELEC.tv/files/7110269/kodi_crashlog_20210903203500.log)
[kodi_crashlog_20210903203906.log](https://github.com/LibreELEC/LibreELEC.tv/files/7110270/kodi_crashlog_20210903203906.log)
[kodi_crashlog_20210903204604.log](https://github.com/LibreELEC/LibreELEC.tv/files/7110271/kodi_crashlog_20210903204604.log)
[kodi_crashlog_20210903215117.log](https://github.com/LibreELEC/LibreELEC.tv/files/7110272/kodi_crashlog_20210903215117.log)

HiassofT, 2021-09-04:
This should already be fixed in our LE10 branch, please test if this build works
https://www.horus.com/~hias/tmp/libreelec/LibreELEC-RPi4.arm-10.0-devel-20210901104638-c91d4d4.tar

so long,

Hias

freynder, 2021-09-04:
Hi @HiassofT ,

Just tested and all seems to work now. Thank you so much!

heitbaum, 2021-09-04:
Closing as fixed

closed 2021-09-04 by heitbaum