LibreELEC/LibreELEC.tv #5612: Chrome works on X11 but on GBM.

heitbaum, 2021-09-08:
```
2021-09-07 21:33:31.002 T:35846    INFO <general>: initializing python engine.
2021-09-07 21:33:34.887 T:35875    INFO <general>: initializing python engine.
2021-09-07 21:33:34.917 T:35875    INFO <general>: CPythonInvoker(33, /storage/.kodi/addons/browser.chrome/default.py): script successfully run
2021-09-07 21:33:34.991 T:35875    INFO <general>: Python interpreter stopped
```

the above log is from GBM - whilst chrome does not actually run. Will need some more investigation (help) to get `/storage/.kodi/addons/browser.chrome/default.py` to run/debug chrome. This PR ok to go for X11 - so suggest we progress this, and work on chrome on GBM as a separate PR.

```
=== tested on ===
X11.x86_64-devel-20210904174934-142782d
Linux nuc11 5.14.1 #1 SMP Sat Sep 4 16:48:42 AEST 2021 x86_64 GNU/Linux
Starting Kodi (19.1 (19.1.0) Git:6402a50950ff6463042f8891ec4029c782730391). Platform: Linux x86 64-bit
Using Release Kodi x64
Kodi compiled 2021-09-04 by GCC 10.3.0 for Linux x86 64-bit version 5.14.1 (331265)
Running on LibreELEC (heitbaum): devel-20210904174934-142782d 11.0, kernel: Linux x86 64-bit version 5.14.1
FFmpeg version/source: 4.4-Kodi
Host CPU: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz, 8 cores available
[    0.000000] DMI: Intel(R) Client Systems NUC11PAHi7/NUC11PABi7, BIOS PATGL357.0040.2021.0414.1645 04/14/2021
CApplication::CreateGUI - using the x11 windowing system
RetroPlayer[PROCESS]: Registering process control for X11
CRenderSystemGL::InitRenderSystem - Version: 4.6 (Core Profile) Mesa 21.2.1, Major: 4, Minor: 6
GL_RENDERER = Mesa Intel(R) Xe Graphics (TGL GT2)
GL_VERSION = 4.6 (Core Profile) Mesa 21.2.1
libva info: VA-API version 1.12.0
vainfo: VA-API version: 1.12 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.3.2 (8aca4af465)
```

_Originally posted by @heitbaum in https://github.com/LibreELEC/LibreELEC.tv/issues/5605#issuecomment-914232549_

lrusak, 2021-09-08:
I'm not really sure what is happening but Kodi will have to be stopped (exited) for Chrome to run. There can only be one DRM master at a time.

heitbaum, 2024-03-16:
Known issue. Closing RFE. If someone want to work on this, please create a new RFE or this can be reopened.

closed 2024-03-16 by heitbaum