LibreELEC/LibreELEC.tv #5635: [BUG] Settings Add-on GUI (Missing Text) problem with Bluetooth Pairing

chewitt, 2021-09-15:
### Describe the bug
This has been reported a few times in the forums. If you pair a device via the current LE10 settings add-on, when bluetoothctl prompts for PIN, the add-on shows a window with no text, so you cannot see the PIN to enter, making it impossible to pair the device. The current workaround is to use bluetoothctl over SSH console.

chewitt, 2021-10-16:
There are three forum reports and I had another on IRC with LE10. Reading the reports again, they are all from users with BT keyboards (where input is possible and thus a PIN is required). I've also seeen this myself with a BT remote with a numeric keypad for input (not a full keyboard). I'd guess it doesn't show up on devices that cannot provide input, which will be the majority of BT devices.

The first report has some pics of the problem (in German). First screen shows clicking 'pair'. Second screens shows the GUI saying "gebe den folgenden code auf deinem gerät.." which means "enter the following code on your device.." but there is no code being displayed. Third pic shows the timeout after not providing the unknown PIN.

https://forum.libreelec.tv/thread/24458-can-t-pair-a-bluetooth-keyboard-with-libreelec-10-and-raspi-4/

https://forum.libreelec.tv/thread/24374-bt-pairing-issue/

https://forum.libreelec.tv/thread/24500-le-10-0-0-adding-a-bluetooth-keyboard-fails-when-pairing-code-is-shown-the-code/

heitbaum, 2022-09-21:
Potentially fixed by https://github.com/LibreELEC/service.libreelec.settings/pull/283

chewitt, 2022-11-29:
Closing as the issue was resolved in 10.0.3 (and master). Thanks!

closed 2022-11-29 by chewitt