LibreELEC/LibreELEC.tv #5646: Lockup of playback/kodi after n plays on repeat.

wakass, 2021-09-16:
### Description
Kodi locks up playback on playlist, after a number of plays on repeat.

### To Reproduce
* Play on repeat in a playlist, an x265,4k file, from the kodi samples page. Enjoy the choreography.. (https://drive.google.com/open?id=0BwxFVkl63-lEdVBuZkltckdZZ0k)
* Causes a freeze after approximately 20 plays.

### Information
* Pi4, 4Gig
* Libreelec 10.0.0

### Log file
I think i've tracked it down to the following errors on .kodi/temp/kodi.log
```2021-09-16 16:39:09.881 T:1219    ERROR <general>: CDVDVideoCodecDRMPRIME::AddData - send packet failed: Unknown error occurred (-1313558101)```

This is concurrent with a dmesg on cma_alloc.
It normally repeats the alloc buf successfully for a number of plays until it bugs out:
```
[   13.632592] rpivid feb10000.codec: rpivid_h265_start: (3840x2160)
......
[ 1171.032737] rpivid feb10000.codec: Alloc buf (6291456) for bit copy OK
[ 1171.080861] rpivid feb10000.codec: Alloc buf (6291456) for bit copy OK
[ 1171.131999] rpivid feb10000.codec: Alloc buf (6291456) for bit copy OK
[ 1474.636251] rpivid feb10000.codec: rpivid_h265_stop
[ 1474.889060] rpivid feb10000.codec: rpivid_h265_start: (3840x2160)
[ 1474.896355] rpivid feb10000.codec: SPS changed
[ 1474.896366] rpivid feb10000.codec: PPS changed
[ 1474.900203] rpivid feb10000.codec: Alloc buf (6291456) for bit copy OK
[ 1474.948713] rpivid feb10000.codec: Alloc buf (6291456) for bit copy OK
[ 1474.998867] rpivid feb10000.codec: Alloc buf (6291456) for bit copy OK
[ 1475.071194] cma: cma_alloc: alloc failed, req-size: 4050 pages, ret: -12
```

popcornmix, 2021-09-17:
Thanks for report. Our codec guy has reproduced a leak in bitstream buffer handling with hevc in a simpler test app. He's investigating the right fix.

popcornmix, 2021-09-22:
An update. The codec guy fixed the bitstream buffer leak and in further testing found another leak (of file handles) that took longer to show.

Now I can run a 10 second version of your file (which fails sooner due to the restarts) in a loop overnight, so several thousand loops.

This should appear in a future build of LE.

wakass, 2021-09-22:
Top notch! Is there a PR/commit/diff I can obtain to compile my own LE and/or affected libraries in the meanwhile?

popcornmix, 2021-09-22:
It's the recent commits here: https://github.com/jc-kynesim/rpi-ffmpeg/tree/test/4.3.2/rpi_main
I expect @HiassofT may PR the updates here when he's happy with testing.

HiassofT, 2021-09-23:
I've PR'ed the ffmpeg update, it should find it's way into the LE10 branch soon

heitbaum, 2021-09-27:
Closing, will find itself into LE 10.0.1 release

closed 2021-09-27 by heitbaum

matiasilva, 2021-11-02:
I know this has been closed already, but I just wanted to provide a second perspective. I had the exact same problem with 10.0.0, and after seeing this issue tested a nightly build -- all working amazing so far.

heitbaum, 2021-11-03:
@matiasilva  - thank you for testing.

PR #5681 (2021-09-23): ffmpeg: update rpi patch
This fixes several memory leaks in H265 HW decode

Fixes: #5646