LibreELEC/LibreELEC.tv #5653: [BUG] Issues passing through Kodi audio to Onkyo Reciever / video to TV, but works perfectly fine from my 2020 Macbook

Jsalas424, 2021-09-17:
### Describe the bug

I recently upgraded my hardware and now cannot get audio from my Kodi to play through my receiver with my video passing through and playing on the TV. I am certain that the receiver is capable of this since I'm perfectly capable of doing this from my Macbook pro.

### To Reproduce
Steps to reproduce the behavior:
1. Hookup this hardware
2. Play anything at all
3. You see video but no audio passes through 

### Informations
 - LE Version: v9.2.8 Kernel: Linux 4.19.127
 - Hardware Platform: - Raspberry Pi 4 8Gb

### Log file

https://paste.kodi.tv/paqiqamico.kodi

### Additional context

Current path is Kodi -> Receiver input -> Receiver output to TV via this HDMI

Hardware details:

- Onkyo TX-NR797

- Raspberry Pi 4 8Gb

- LibreELEC v9.2.8 Kernel: Linux 4.19.127

- Kodi v18.9.0

- StarTech 3ft Micro HDMI to HDMI Cable with Ethernet

I've attempted to set this up on Kodi with and without audio passthrough. Here are my current settings: 
[image]

2020 MacBook Pro can do this fine using this cable: https://www.amazon.com/gp/product/B075V5JK36/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 & I've managed to do this using a very regular HDMI + this tiny monitor

Thanks for the help!

heitbaum, 2022-02-17:
@Jsalas424   Is this still an issue with LE10/11? Is so please reopen.

closed 2022-02-17 by heitbaum

Jsalas424, 2022-02-17:
> @Jsalas424 Is this still an issue with LE10/11? Is so please reopen.

I gave up on this and repurposed that Pi as my homeassistant. Sorry for the disappointing news but I'll be certain to report back here if I ever revisit this effort.