LibreELEC/LibreELEC.tv #5671: [BUG] Kodi breaks CEC after first power cycle

NetDwarf, 2021-09-20:
### Describe the bug
After first power cycle CEC stops working on most TVs and probably all LE devices (more on tested models in Informations).

### To Reproduce
Steps to reproduce the behavior:

0. (Fresh install)
1. Power LE device and TV on (CEC works here)
2. Turn down TV
3. Turn on TV

(CEC stops working)

### Informations
LE devices tested: Rock64 (this is the one I am focussing on) and Q Plus which works with Tanix TX6 img. I am rather sure I had that same thing with RPi3 on nightlies and expect it to have this error as well, but I do not have access to one currently.

TVs where it exhibits this bug and were okay with LE9:
- LG 55LE5300
- Sharp LC40FE4042E
- Samsung LE37A615
- Samsung LE32D579
- Toshiba 55WL863

I also tested a newer Samsung (I would need to ask for the exact model) which worked, even if it took some time to connect after turning on TV again. 

### Log file
`cec-ctl -S` before powering down (CEC working)
```
Driver Info:
	Driver Name                : dwhdmi-rockchip
	Adapter Name               : dw_hdmi
	Capabilities               : 0x0000010e
		Logical Addresses
		Transmit
		Passthrough
		Connector Info
	Driver version             : 5.10.47
	Available Logical Addresses: 4
	DRM Connector Info         : card 0, connector 51
	Physical Address           : 1.0.0.0
	Logical Address Mask       : 0x0002
	CEC Version                : 1.4
	Vendor ID                  : 0x001582 (Pulse-Eight)
	OSD Name                   : '' 
	Logical Addresses          : 1 (Allow Fallback to Unregistered)

	  Logical Address          : 1 (Recording Device 1)
	    Primary Device Type    : Record
	    Logical Address Type   : Record

	System Information for device 0 (TV) from device 1 (Recording Device 1):
		CEC Version                : 1.4
		Physical Address           : 0.0.0.0
		Primary Device Type        : TV
		Vendor ID                  : 0x000039 (Toshiba)
		OSD Name                   : 'TOSHIBA TV'
		Menu Language              : eng
		Power Status               : On

	Topology:

	0.0.0.0: TV
	    1.0.0.0: Recording Device 1
```

`cec-ctl -S` after power cycle. (CEC dead)
```
Driver Info:
	Driver Name                : dwhdmi-rockchip
	Adapter Name               : dw_hdmi
	Capabilities               : 0x0000010e
		Logical Addresses
		Transmit
		Passthrough
		Connector Info
	Driver version             : 5.10.47
	Available Logical Addresses: 4
	DRM Connector Info         : card 0, connector 51
	Physical Address           : f.f.f.f
	Logical Address Mask       : 0x0000
	CEC Version                : 1.4
	Vendor ID                  : 0x001582 (Pulse-Eight)
	OSD Name                   : ''
	Logical Addresses          : 1 (Allow Fallback to Unregistered)

	  Logical Address          : Not Allocated
	    Primary Device Type    : Record
	    Logical Address Type   : Record

	System Information for device 0 (TV) from device 15 (Unregistered):
		CEC Version                : 
		Physical Address           : 
		Vendor ID                  : 
		OSD Name                   : 

	Topology:
```

`systemctl stop kodi && cec-client` (**CEC kept working after power cycle!**):
```
TRAFFIC: [           31721]	>> 0f:36
DEBUG:   [           31721]	TV (0): power status changed from 'on' to 'standby'
DEBUG:   [           31721]	>> TV (0) -> Broadcast (F): standby (36)
DEBUG:   [           43402]	CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0000 phys_addr=ffff
DEBUG:   [           44320]	CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0000 phys_addr=1000
DEBUG:   [           44461]	CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=0002 phys_addr=1000
DEBUG:   [           44812]	CLinuxCECAdapterCommunication::Process - ioctl CEC_RECEIVE - rx_status=01 len=2 addr=01 opcode=46
TRAFFIC: [           44812]	>> 01:46
DEBUG:   [           44812]	<< Recorder 1 (1) -> TV (0): OSD name 'CECTester'
TRAFFIC: [           44812]	<< 10:47:43:45:43:54:65:73:74:65:72
DEBUG:   [           44812]	>> TV (0) -> Recorder 1 (1): give osd name (46)
DEBUG:   [           45096]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=11 addr=10 opcode=47
DEBUG:   [           45163]	CLinuxCECAdapterCommunication::Process - ioctl CEC_RECEIVE - rx_status=01 len=2 addr=01 opcode=c3
TRAFFIC: [           45163]	>> 01:c3
DEBUG:   [           45163]	sending abort with opcode c3 and reason 'unrecognised opcode' to TV
DEBUG:   [           45163]	<< transmitting abort message
TRAFFIC: [           45163]	<< 10:00:c3:00
DEBUG:   [           45163]	>> TV (0) -> Recorder 1 (1): request ARC start (C3)
DEBUG:   [           45276]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT - tx_status=01 len=4 addr=10 opcode=00
```

### Additional context

As you can tell from the logs; restarting kodi enables CEC once again, but otherwise it's dead.

jernejsk, 2021-09-25:
By default, when you power off your TV, your SBC/box should also turn off (TV gives notification via CEC). Did you disable that?

Anyway, if I disable that setting, Tanix TX6 connected to LG B8 works as expected - CEC is still functional once TV is turned on again.

Based on the fact that `cec-ctl -S` shows non-responsive interface, it's much more likely that this is a driver issue. Rockchip images switched from vendor to mainline kernel between LE9 and LE10. Note that CEC is known to be quirky and it's hard to fix anything without a setup at hand which reliably reproduces the issue.

NetDwarf, 2021-09-26:
> By default, when you power off your TV, your SBC/box should also turn off (TV gives notification via CEC). Did you disable that?

I did not do that on this particular test (but on my regular setup), but repeating it with this setting makes no difference. Actually two things are of notice here: 
1. It does not put the device in "standby" (which is probably intended behavior, since Rock64 does not support it).
2. CEC breaks on TV start, because I can wake the TV from sleep by deactivating screensaver and then it's broken

> Anyway, if I disable that setting, Tanix TX6 connected to LG B8 works as expected - CEC is still functional once TV is turned on again.

Yeah I suspect every new 4K TV to be working as I eluded before, that new Samsung does work (tho CEC seems to reconnect when turning on TV again, which is maybe associated with this). If you have an "old" TV from say >5 years I suspect the chances are higher that you can reproduce it. 

> Based on the fact that cec-ctl -S shows non-responsive interface, it's much more likely that this is a driver issue or at least combination of Kodi interraction with the driver. Rockchip images switched from vendor to mainline kernel between LE9 and LE10. Note that CEC is known to be quirky and it's hard to fix anything without a setup at hand which reliably reproduces the issue.

Yeah, that is always hard to do. That being said, if you find a TV where this happens (CEC works and after TV power cycle it doesn't you can reliably reproduce it on every restart of Kodi as in `systemctl restart kodi` -> works -> TV power cycle -> doesn't work).

jernejsk, 2021-09-26:
I only have this TV. I'll test in a couple of weeks, when I'll have access to an old plasma TV over weekend.

Note that 4K TVs most likely support HDMI 2.0 (or newer) and FullHD TVs supports HDMI 1.4 (or older). There were some slight changes in CEC protocol between 1.4 and 2.0. I have to check if one of these changes needs to be handled differently.

knaerzche, 2021-10-15:
We will update this as soon as something has changed

Main problem here is: Neither  @jernejsk nor I can reproduce this issue with our TVs and its very hard to guess where it comes from without actually having this issue.

jernejsk, 2021-10-16:
Sorry, I was unable to test on older one yet (it's not at my place).

knaerzche, 2021-10-16:
@NetDwarf 
Today I could verify your issue with Rock64 on my mother's TV and think I could also fix it (at least for her TV)
Please download https://mega.nz/file/2HZ00DDT#O7eB6HJM-5UM_M8CSg4DcTLE-FRw_fBKMWJt_s2SOwM and flash it to a spare SD card (note: it's based on current LE master, so there might be some (other) bugs)

While switching TV on/off please be logged in your in your device and watch `cec-ctl -m` (don't change any CEC settings within kodi)
After switching off you should see something like
`Received from TV to all (0 to 15): STANDBY (0x36)`

After switching the TV on again you should see this (note: I can take serveral seconds after TV is switched on again, before messages appear)
```
Received from TV to all (0 to 15): ROUTING_CHANGE (0x80):
	orig-phys-addr: 0.0.0.0
	new-phys-addr: 4.0.0.0
Transmitted by Recording Device 1 to TV (1 to 0): IMAGE_VIEW_ON (0x04)
Received from TV to all (0 to 15): REQUEST_ACTIVE_SOURCE (0x85)
Received from TV to all (0 to 15): ROUTING_CHANGE (0x80):
	orig-phys-addr: 0.0.0.0
	new-phys-addr: 4.0.0.0
Transmitted by Recording Device 1 to all (1 to 15): ACTIVE_SOURCE (0x82):
	phys-addr: 4.0.0.0
	Tx, Low Drive (4), Max Retries
Received from TV to Recording Device 1 (0 to 1): POLL
Transmitted by Recording Device 1 to all (1 to 15): ACTIVE_SOURCE (0x82):
	phys-addr: 4.0.0.0
Transmitted by Recording Device 1 to TV (1 to 0): MENU_STATUS (0x8e):
	menu-state: activated (0x00)
Received from TV to Recording Device 1 (0 to 1): MENU_REQUEST (0x8d):
	menu-req: query (0x02)
Received from TV to Recording Device 1 (0 to 1): GIVE_DECK_STATUS (0x1a):
	status-req: on (0x01)
Transmitted by Recording Device 1 to all (1 to 15): ACTIVE_SOURCE (0x82):
	phys-addr: 4.0.0.0
	Tx, Low Drive (4), Max Retries
....

Transmitted by Recording Device 1 to TV (1 to 0): MENU_STATUS (0x8e):
	menu-state: activated (0x00)
```
After you've seen the last message (or a similar one) CEC should be usable again. (no, you don't have to run `cec-ctl -m` all the time, it just for verification ;)

Please test with your Rock64 and since you seem to have lots of TVs available, it would be great to know, if it fixes it for any and/or for all.

NetDwarf, 2021-10-17:
> Other then you suspected this does not relate to kodi or libcec - it's an issue in the hdmi driver which is used by Rockchip and Allwinner (and Amlogic).
> If the issue exists on RPi also, it's likely related to their hdmi driver.

That is very interesting! I am going to make sure if it exists on RPi3/4 then regardless. Thank you for the clarification :slightly_smiling_face:

NetDwarf, 2021-10-17:
Okay I tested RPi3 and RPi4:

RPi3 on LibreELEC-RPi2.arm-11.0-nightly-20211017-d3745e7.img.gz: **Works!**

RPi4 on LE10.0: **Works!**

Okay, so it was probably broken at some point, but is now definitely fixed. Only thing I noticed here as well is the long delay until it works. It's something different and if it doesn't cause any other issues then it's rather minor.

(Update OP accordinlgy)

NetDwarf, 2021-10-20:
Okay, back home and immediately tested the fix and it works. :partying_face: 

`cec-ctl -m` output when switching TV off:

```
[...]
Received from TV to all (0 to 15): STANDBY (0x36)
Received from TV to Recording Device 1 (0 to 1): POLL
[...]
```

`cec-ctl -m` output when switching TV on:

```
[...]
Received from TV to all (0 to 15): ROUTING_CHANGE (0x80):
	orig-phys-addr: 0.0.0.0
	new-phys-addr: 1.0.0.0
Transmitted by Recording Device 1 to TV (1 to 0): IMAGE_VIEW_ON (0x04)
Received from TV to Recording Device 1 (0 to 1): REQUEST_ARC_INITIATION (0xc3)
Transmitted by Recording Device 1 to all (1 to 15): ACTIVE_SOURCE (0x82):
	phys-addr: 1.0.0.0
Transmitted by Recording Device 1 to TV (1 to 0): FEATURE_ABORT (0x00):
	abort-msg: 195 (0xc3, REQUEST_ARC_INITIATION)
	reason: unrecognized-op (0x00)
Transmitted by Recording Device 1 to TV (1 to 0): IMAGE_VIEW_ON (0x04)
Transmitted by Recording Device 1 to all (1 to 15): ACTIVE_SOURCE (0x82):
	phys-addr: 1.0.0.0
Received from TV to Recording Device 1 (0 to 1): POLL
[...]
```

It's mostly as you (@knaerzche) said, except one feature was missing, which is a limitation of the TV (or port) I guess. It would be nice to know if this fix is also portable to Allwinner H6 devices (Tanix TX6) and LE10, but thank you anyway. Awesome work!