LibreELEC/LibreELEC.tv #5675: [BUG] glibc-2.32 and later has a regression with lchmod when /proc is not mounted (such as a chroot)

wagnerch, 2021-09-22:
### Describe the bug
rsync specifically was the reported issue, which by default when running in "--daemon" mode has "use chroot" set to "yes".  So the rsync daemon will run inside the chroot for the configured storage path, and when pushing to LE 10 the permissions fail with:
rsync: [receiver] failed to set permissions on "/Synology3_1/.synobkpinfo.db.Di1ntS" (in Synology3_Backup): Operation not supported (95)
 
See https://forum.libreelec.tv/thread/24557-rsync-not-working-anymore-after-upgrading-to-le10/ for the discussion.

Upstream bug is glibc:
https://sourceware.org/bugzilla/show_bug.cgi?id=26401

rsync has applied a workaround for the glibc bug:
https://github.com/WayneD/rsync/issues/109

Arch is cherry picking these 2 commits on top of rsync 3.2.3, it seems:

# Force HAVE_LCHMOD off for Linux (for now).
git cherry-pick -n 85b8dc8abaca96fc3ea7421e09101b6ac41b6718
# Work around glibc's lchmod() issue a better way.
git cherry-pick -n 9dd62525f3b98d692e031f22c02be8f775966503

### To Reproduce
Steps to reproduce the behavior:
1. Using rsync locally inside a chroot without /proc mounted: https://github.com/WayneD/rsync/issues/109#issue-728110332
2. Using a C program to call lchmod to change permissions: https://github.com/WayneD/rsync/issues/109#issuecomment-718919262

### Informations
 - LE Version: 10.0.0

### Log file
https://forum.libreelec.tv/core/attachment/7864-rsync-log-le10-txt/

### Additional context

heitbaum, 2021-09-23:
Just reached out to the rsync list to see if there is an imminent rsync release. https://lists.samba.org/archive/rsync/2021-September/032678.html  If not - then suggest we proceed.

mglae, 2021-09-23:
There is a second minor issue reported in the forum: rsync is showing version 8.95.002-6436-g4b15caf9.

Please add 
```
diff --git a/packages/network/rsync/package.mk b/packages/network/rsync/package.mk
index dad8087252..0837052c33 100644
--- a/packages/network/rsync/package.mk
+++ b/packages/network/rsync/package.mk
@@ -30,3 +30,12 @@ PKG_CONFIGURE_OPTS_TARGET="--disable-acl-support \
                            --disable-zstd \
                            --with-included-popt \
                            --without-included-zlib"
+
+pre_make_host() {
+  # do not detect LE git version
+  echo "#define RSYNC_GITVER \"${PKG_VERSION}\"" >git-version.h
+}
+
+pre_make_target() {
+  pre_make_host
+}
```
to the PR.

wagnerch, 2021-09-23:
@mglae I was thinking about the same, can add it to the PRs.

PR #5676 (2021-10-02): [LE11] rsync: add lchmod workaround patches
Fixes #5675, which is a regression introduced by glibc implementation of lchmod.  It appears that glibc is more or less waiting on a new syscall from the kernel to fix it properly, until then lchmod is broken when /proc is not mounted.

Tested on LE10 using rsync.conf:
```
list = yes
port = 873
log file = /storage/rsyncd.log
timeout = 300

[Synology3_Backup]
comment = Synology3 Video Backup
path = /tmp/test
max connections = 0
read only = false
hosts allow = 192.168.0.1
uid = root
gid = root
```

And running rsync as daemon:
```
mkdir -p /tmp/test
rs [... cut]
PR #5677 (2021-10-02): [LE10] rsync: add lchmod workaround patches
Fixes #5675, which is a regression introduced by glibc implementation of lchmod.  It appears that glibc is more or less waiting on a new syscall from the kernel to fix it properly, until then lchmod is broken when /proc is not mounted.

Tested on LE10 using rsync.conf:
```
list = yes
port = 873
log file = /storage/rsyncd.log
timeout = 300

[Synology3_Backup]
comment = Synology3 Video Backup
path = /tmp/test
max connections = 0
read only = false
hosts allow = 192.168.0.1
uid = root
gid = root
```

And running rsync as daemon:
```
mkdir -p /tmp/test
rs [... cut]
closed 2021-10-02 by CvH