LibreELEC/LibreELEC.tv #5678: [Require]What is the frame rate using hantro VPU?

hptsf, 2021-09-23:
Hello，

What is the frame rate of the test?

In my case is： it up to more than 110fps, I am a little afraid to believe it。
I using kernel5.4, and apply the https://github.com/LibreELEC/LibreELEC.tv/tree/master/projects/Rockchip/patches/linux/default patches。
testing command using： -hwaccel drm

frame= 1259 fps=118 q=-0.0 Lsize=N/A time=00:00:42.04 bitrate=N/A speed=3.93x    
video:659kB audio:7884kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown

and the video‘s format is:
[graph 0 input from stream 0:0 @ 0xaaaabcd88a60] Setting 'video_size' to value '3840x2160'
[graph 0 input from stream 0:0 @ 0xaaaabcd88a60] Setting 'pix_fmt' to value '23'
[graph 0 input from stream 0:0 @ 0xaaaabcd88a60] Setting 'time_base' to value '1/29970'
[graph 0 input from stream 0:0 @ 0xaaaabcd88a60] Setting 'pixel_aspect' to value '1/1'
[graph 0 input from stream 0:0 @ 0xaaaabcd88a60] Setting 'sws_param' to value 'flags=2'
[graph 0 input from stream 0:0 @ 0xaaaabcd88a60] Setting 'frame_rate' to value '2997/100'
[graph 0 input from stream 0:0 @ 0xaaaabcd88a60] w:3840 h:2160 pixfmt:nv12 tb:1/29970 fr:2997/100 sar:1/1 sws_param:flags=2

jernejsk, 2021-09-24:
Questions should be posted in appropriate section on forum, issue tracker is for LE issues only.

closed 2021-09-24 by jernejsk

hptsf, 2021-09-26:
what's the web address of forum?

heitbaum, 2021-09-26:
> what's the web address of forum?
Forum link on https://libreelec.tv/