LibreELEC/LibreELEC.tv #5715: Building fails in glib/libglib-2.0.a.p/gspawn.c.o

coffeinflash, 2021-10-04:
I am trying to build the latest master branch on openSUISE Tumbleweed, but it failes with following error:

`Option buildtype is: plain [default: debugoptimized]
Found ninja-1.10.2 at /home/Herbie/source/LibreELEC.tv/build.LibreELEC-RPi2.arm-11.0-devel/toolchain/bin/ninja
[1/350] Linking static target glib/pcre/libpcre.a
[2/350] Compiling C object gobject/libgobject-2.0.a.p/gatomicarray.c.o
[3/350] Compiling C object glib/libglib-2.0.a.p/gspawn.c.o
FAILED: glib/libglib-2.0.a.p/gspawn.c.o 
/home/Herbie/source/LibreELEC.tv/build.LibreELEC-RPi2.arm-11.0-devel/toolchain/bin/host-gcc -Iglib/libglib-2.0.a.p -Iglib -I../glib -I. -I.. -I/home/Herbie/source/LibreELEC.tv/build.LibreELEC-RPi2.arm-11.0-devel/toolchain/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -D_GNU_SOURCE -fno-strict-aliasing -Wduplicated-branches -Wimplicit-fallthrough -Wmisleading-indentation -Wstrict-prototypes -Wunused -Wno-unused-parameter -Wno-bad-function-cast -Wno-cast-function-type -Wno-pedantic -Wno-format-zero-length -Werror=declaration-after-statement -Werror=format=2 -Werror=implicit-function-declaration -Werror=init-self -Werror=missing-include-dirs -Werror=missing-prototypes -Werror=pointer-arith -march=native -O2 -Wall -pipe -Wno-format-security -fPIC -pthread '-DG_LOG_DOMAIN="GLib"' -DGLIB_COMPILATION -MD -MQ glib/libglib-2.0.a.p/gspawn.c.o -MF glib/libglib-2.0.a.p/gspawn.c.o.d -o glib/libglib-2.0.a.p/gspawn.c.o -c ../glib/gspawn.c
../glib/gspawn.c: In function 'safe_closefrom':
../glib/gspawn.c:1497:7: error: too few arguments to function 'close_range'
1497 |   if (close_range (lowfd, G_MAXUINT) != 0 && errno == ENOSYS)
|       ^~~~~~~~~~~
In file included from /usr/include/unistd.h:1205,
from /usr/include/bits/sigstksz.h:24,
from /usr/include/signal.h:328,
from /usr/include/sys/wait.h:36,
from ../glib/gspawn.c:25:
/usr/include/bits/unistd_ext.h:56:12: note: declared here
56 | extern int close_range (unsigned int __fd, unsigned int __max_fd,
|            ^~~~~~~~~~~
[4/350] Compiling C object gobject/libgobject-2.0.a.p/genums.c.o
[5/350] Compiling C object gobject/libgobject-2.0.a.p/gboxed.c.o
[6/350] Compiling C object gobject/libgobject-2.0.a.p/gmarshal.c.o
[7/350] Compiling C object gobject/libgobject-2.0.a.p/gbinding.c.o
[8/350] Generating glib_enumtypes_c with a custom command (wrapped by meson to capture output)
[9/350] Compiling C object gobject/libgobject-2.0.a.p/gclosure.c.o
[10/350] Compiling C object gobject/libgobject-2.0.a.p/gobject.c.o
ninja: build stopped: subcommand failed.
FAILURE: scripts/build glib:host during make_host (default)
*********** FAILED COMMAND ***********
ninja ${NINJA_OPTS} ${PKG_MAKE_OPTS_HOST}
**************************************
FAILURE: scripts/build glib:host has failed!

The following log for this failure is available:
/home/Herbie/source/LibreELEC.tv/build.LibreELEC-RPi2.arm-11.0-devel/.threads/logs/55.log

>>> glib:host seq 55 >>>
[053/275] [FAIL] build   glib:host`

[55.log](https://github.com/LibreELEC/LibreELEC.tv/files/7275628/55.log)

heitbaum, 2021-10-04:
You may want to try with: #5716 - this is what I am currently building with, does enable compile with glibc2.34 and fixed the namespace clashes.

https://lists.buildroot.org/pipermail/buildroot/2021-September/622832.html

coffeinflash, 2021-10-05:
Thx, building was successful now with the updated glib.