LibreELEC/LibreELEC.tv #5844: [DEVELOPMENT] libgpg-error 1.43 picking up wrong library version

heitbaum, 2021-11-03:
### Describe the bug
Whilst I had reverted my bump of libgpg-error 1.42 in my dev tree, 1.43 was released today - and  fails too. On a quick look - it sems that it is picking up version 1.41 from somewhere. I have not done any further analysis at this time. Placeholder to investigate - as there were changes in 1.42+ to better support cross-compiling.

```
diff --git a/packages/security/libgpg-error/package.mk b/packages/security/libgpg-error/package.mk
index 14c9b1ebc3..777230b763 100644
--- a/packages/security/libgpg-error/package.mk
+++ b/packages/security/libgpg-error/package.mk
@@ -3,8 +3,8 @@
 # Copyright (C) 2018-present Team LibreELEC (https://libreelec.tv)

 PKG_NAME="libgpg-error"
-PKG_VERSION="1.41"
-PKG_SHA256="64b078b45ac3c3003d7e352a5e05318880a5778c42331ce1ef33d1a0d9922742"
+PKG_VERSION="1.43"
+PKG_SHA256="a9ab83ca7acc442a5bd846a75b920285ff79bdb4e3d34aa382be88ed2c3aebaf"
 PKG_LICENSE="GPLv2"
 PKG_SITE="https://www.gnupg.org"
 PKG_URL="https://www.gnupg.org/ftp/gcrypt/libgpg-error/${PKG_NAME}-${PKG_VERSION}.tar.bz2"
```

```
$ cat build.LibreELEC-Generic.x86_64-11.0-devel/build/libgpg-error-1.43/.x86_64-libreelec-linux-gnu/src/gpg-error-config-test.log
Test result: --version
====================: ./gpg-error-config-old
1.43
====================: gpgrt-config
1.41
====================
Test result: --mt --libs
====================: ./gpg-error-config-old
-lgpg-error
====================: gpgrt-config
-lgpg-error -lpthread
====================
Test result: --mt --cflags --libs
====================: ./gpg-error-config-old
-lgpg-error
====================: gpgrt-config
-lgpg-error -lpthread
====================
Test result: --variable=mtlibs
====================: ./gpg-error-config-old

====================: gpgrt-config
-lpthread
====================
Test result: --version
====================: ./gpg-error-config-old
1.43
====================: gpgrt-config
1.41
====================
```

mglae, 2021-11-06:
Did you perform a clean build?

heitbaum, 2021-11-06:
@mglae - thank you - was as simple as that 👍

closed 2021-11-06 by heitbaum