LibreELEC/LibreELEC.tv #5858: [RK3399] HDMI Passthrough not working (or shown)

zeus24, 2021-11-07:
Hi,
I have problems with the HDMI passthrough in the lastest Kodi nightly. The setting "allow passthrough" is not shown in the Kodi setup and also not the setting, where I can select what my receiver is capable off (AC3, EAC3, Dolby Plus etc). 

HW:
- LG OLED
- Yamaha Receiver V477
- Rockpro64
(Rockpro64 connected via HDMI to the Receiver)

Logs: 
http://ix.io/3Ei5

The HDMI is correctly  recognized in aplay and I also see it in the KODI log. 
I already tried a few different things to get it working, e.g. change HDMI cable or connection etc. As a final solution I "just" tried a different distro ([Recalbox](https://github.com/mrfixit2001/recalbox-rockchip/releases), Kernel 5.7 with Kodi 18.5) and there it worked out of the box.

If anyone can give me a hint, under what circumstances the settings are hidden in Kodi (EDID? CEC?) I could also try to further investigate, but at the moment I am out of ideas. 

Device 3
2021-11-07 12:10:05.417 T:957      INFO <general>:         m_deviceName      : hdmi:CARD=HDMI,DEV=0
2021-11-07 12:10:05.417 T:957      INFO <general>:         m_displayName     : HDMI
2021-11-07 12:10:05.417 T:957      INFO <general>:         m_displayNameExtra: GSM LG TV SSCR2 on HDMI
2021-11-07 12:10:05.417 T:957      INFO <general>:         m_deviceType      : AE_DEVTYPE_HDMI
2021-11-07 12:10:05.417 T:957      INFO <general>:         m_channels        : FL, FR, LFE, FC, BL, BR, BC, BLOC, BROC
2021-11-07 12:10:05.417 T:957      INFO <general>:         m_sampleRates     : 32000,44100,48000,88200,96000,176400,192000
2021-11-07 12:10:05.417 T:957      INFO <general>:         m_dataFormats     : AE_FMT_RAW,AE_FMT_S24NE4,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S16LE,AE_FMT_RAW
2021-11-07 12:10:05.417 T:957      INFO <general>:         m_streamTypes     : STREAM_TYPE_AC3,STREAM_TYPE_DTSHD,STREAM_TYPE_DTSHD_MA,STREAM_TYPE_DTSHD_CORE,STREAM_TYPE_DTS_1024,STREAM_TYPE_DTS_2048,STREAM_TYPE_DTS_512,STREAM_TYPE_EAC3,STREAM_TYPE_TRUEHD

jernejsk, 2021-11-07:
HDMI passthrough option is hidden intentionally, because driver support is missing.

fliegenderfish, 2022-05-21:
Could someone confirm that this project is dead? Thats what it seems to me!
Thanks

knaerzche, 2022-09-12:
As passthrough is currently not supported by the upstream hdmi driver *that* is certainly not the place to complain about this.
There are members of LE team (including myself) who are working on hw drivers but we usually do that in personal interest, with personal priorities and whenever time allows.

Please report to a kernel mailing list.

closed 2022-09-12 by knaerzche

avi951, 2025-02-28:
Any update on this issue?

rah2501, 2025-03-17:
> Any update on this issue?

I had a chat with the people in the #linux-rockchip IRC channel (on Libera) and apparently nobody is working on passthrough support.