LibreELEC/LibreELEC.tv #5860: Libreelec nightly not starting on amd rx 6600 xt

e-ru, 2021-11-07:
### Describe the bug
Libreelec get stuck at boot screen

[image]

### To Reproduce
Steps to reproduce the behavior:
1. Create a boot stick with the official libreelec usb creation tool 
2. Press enter on select screen ( `i`, `r` and `l` leads to following message: `failed: no such file or directory` )
3. install Libreelec on other disk
4. reboot
5. stuck at boot screen (see first image)

### Informations
 - LE Version: nightly 20211021-45d6560
 - Hardware Platform: x86-64, cpu: intel 11400, gpu: rx 6600 xt

### Log file
no logs due to stuck at boot screen

### Additional context
-

lrusak, 2021-11-07:
remove `quiet` from the syslinux.cfg or try adding `ssh` to allow ssh access by default. Then you should be able to see more output and/or get logs via ssh.

e-ru, 2021-11-11:
hi

thank you very much for your response and sorry for the late reply (need to rearrange some network components... )

the boot process gets stuck after `Reached target System Time synchronized` (see picture)
[image]

here are the logs
[nightly-logs](http://ix.io/3EJc)

if you need more, just let me know

kind regards
erik

lrusak, 2021-11-11:
Hmm, seems that the drm subsystem is failing to initialize and Kodi is running in headless gbm mode.

This seems odd
```
2021-11-11 20:42:24.370 T:875      INFO <general>: GL_VENDOR = Intel
2021-11-11 20:42:24.370 T:875      INFO <general>: GL_RENDERER = Mesa Intel(R) Graphics (RKL GT1)

```

I assume you have a discrete graphics card installed?

Where did you get this LibreELEC build from? Is it built with AMD drivers?

e-ru, 2021-11-12:
hi

Yes, the display is connected to a dedicated amd rx 6600xt.

[here](https://test.libreelec.tv/LibreELEC-Generic.x86_64-11.0-nightly-20211112-b431638.img.gz) i get the image from. don't know if it is built with the latest amd drivers (i guess it's `amdgpu`).

i assume that the nightlys builds are made from the master branch so the kernel version is `5.15` (refering to [this](https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/linux/package.mk) ). meaning that the driver `amdgpu` is recent enough to support the mentioned card.

thank you
erik

lrusak, 2021-11-12:
Can you run the 'modetest' utility and post the output to pastebin?

mglae, 2021-11-12:
There is at least one firmware missing:
```
Nov 11 20:42:22.797108 LibreELEC kernel: amdgpu 0000:03:00.0: Direct firmware load for amdgpu/dimgrey_cavefish_sos.bin failed with error -2
Nov 11 20:42:22.797137 LibreELEC kernel: amdgpu 0000:03:00.0: amdgpu: failed to init sos firmware
Nov 11 20:42:22.797142 LibreELEC kernel: [drm:psp_sw_init] *ERROR* Failed to load psp firmware!
Nov 11 20:42:22.797146 LibreELEC kernel: [drm:amdgpu_device_init.cold] *ERROR* sw_init of IP block <psp> failed -2
Nov 11 20:42:22.797174 LibreELEC kernel: amdgpu 0000:03:00.0: amdgpu: amdgpu_device_ip_init failed
Nov 11 20:42:22.797204 LibreELEC kernel: amdgpu 0000:03:00.0: amdgpu: Fatal error during GPU init
Nov 11 20:42:22.797233 LibreELEC kernel: amdgpu 0000:03:00.0: amdgpu: amdgpu: finishing device.
Nov 11 20:42:22.797239 LibreELEC kernel: amdgpu: probe of 0000:03:00.0 failed with error -2
```

You can get the firmware (and possible others) from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/amdgpu and store it into /storage/.config/firmware/amdgpu

In addition the Intel GPU is detected and used. This can be disabled with the `initcall_blacklist=i915_init` kernel parameter.

e-ru, 2021-11-14:
hello

thank you very much for your help. i tried the nightly from the14th of nov and it's working just fine.

Again, thank you very much
erik

closed 2021-11-14 by e-ru