LibreELEC/LibreELEC.tv #5866: [BUG] Realtek Bluetooth Problem

warmaster668, 2021-11-09:
### Describe the bug
Bluetooth device disconnects at random and it cannot connect until reboot and the problem will start again after connection to bluetooth device and the internet connection will stop working when connected to bluetooth device and after disconnection of bluetooth device will make the internet connection start working again as usual

### To Reproduce
Steps to reproduce the behavior:
1. Go to LibreELEC Settings and Bluetooth option
2. Try to connect to any Bluetooth device that it finds and wait for disconnection and try to go to any internet connected addon
3. see the problem take effect

### Informations
 - LE Version: 10.0.0,10.0.1
 - Hardware Platform: x86_64

### Additional context
LibreELEC 10.0.0,10.0.1
Kodi 19.1,19.3
Realtek Bluetooth 4.2 Adapter
HP Slimline Desktop S01-pf1013w

heitbaum, 2021-11-09:
was #5778

heitbaum, 2022-02-17:
@warmaster668 is this still happening with LE nightlies (LE11) - a number of updates have been added since. Please reopen if this is still an issue. Updates have been made to the LivreElec settings too in both LE10 and LE11 branches.

closed 2022-02-17 by heitbaum