LibreELEC/LibreELEC.tv #5868: [BUG] current master cannot build "slang" due to missing pcre.h

sqozz, 2021-11-10:
### Describe the bug
I'm trying to build the current master (commit abfe01743def33004975d118969886142d063020) for generic x86_64. I use `make release` to build it on gentoo. While building the slang package it fails repeatedly because it cannot find `pcre.h`.

### To Reproduce
Steps to reproduce the behavior:
```
git clone https://github.com/LibreELEC/LibreELEC.tv.git
cd LibreELEC.tv
make release
```

### Information
 - LE Version: current master (commit abfe01743def33004975d118969886142d063020)
 - Hardware Platform: x86_64
 - Build Host: gentoo
 - GCC Version: gcc (Gentoo 9.3.0-r2 p4) 9.3.0

### Log file
[247.log](https://github.com/LibreELEC/LibreELEC.tv/files/7513129/247.log)

### Additional context
I'm pretty sure the `pcre.h` file should be provided by the build system. However just to be sure I also checked that the build host has it installed (dev-libs/libpcre-8.45). The file is present in /usr/include/pcre.h

heitbaum, 2021-11-10:
pcre.h should be there. There is an issue with the PKG_URL of pure #5859 - but I don’t think that is your issue. Here are the pcre.h in my build tree.

```
LibreELEC.tv$ find build.LibreELEC-Generic.x86_64-11.0-devel/ -name pcre.h
build.LibreELEC-Generic.x86_64-11.0-devel/build/pcre-8.45/.x86_64-linux-gnu/pcre.h
build.LibreELEC-Generic.x86_64-11.0-devel/build/pcre-8.45/.x86_64-libreelec-linux-gnu/pcre.h
build.LibreELEC-Generic.x86_64-11.0-devel/install_pkg/pcre-8.45/usr/include/pcre.h
build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/include/pcre.h
build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/pcre.h
```

sqozz, 2021-11-10:
Interesting, on my system nothing is found:
```
sqozz@workstation ~/LibreELEC.tv ‹master› » find build.LibreELEC-Generic.x86_64-11.0-devel -name pcre.h
sqozz@workstation ~/LibreELEC.tv ‹master› » echo $?
0
```

could it be caused by me running `make -j17 release` previously and some kind of out-of-order build?

heitbaum, 2021-11-10:
Try the build, clean then the build of pcre.

heitbaum, 2021-11-10:
I was able to reproduce the error. PR raised.

closed 2021-11-13 by CvH