LibreELEC/LibreELEC.tv #5875: [BUG] Unable to edit any configuration files from SSH

Eshanepicfighter, 2021-11-15:
### Describe the bug
Unable to edit any configuration files via an SSH connection. Even after remounting with read and write permissions, I cannot edit any file in the /etc directory.

### To Reproduce
Steps to reproduce the behavior:
1. SSH into a Libreelec Device
2. Remount the / directory with read and write permissions
3. CD into the /etc/systemd/ directory
4. Edit the timesyncd.conf file and try saving changes

### Informations
 - LE Version: [e.g. 9.2.1]
 - Hardware Platform: [e.g. RPi3]

### Log file

### Additional context

antonlacon, 2021-11-16:
This isn't a bug; write permissions still won't allow changes to a read-only squashfs image.

You can make your intended changes in `/etc/system.d/timesyncd.conf.d/` in `/storage/.config/timesyncd.conf.d/` instead. If you have additional support questions, please use the forums: https://forum.libreelec.tv/

Eshanepicfighter, 2021-11-16:
Sorry to continue the issue but I'm having some difficulties using the LibreELEC forum. 

Would I just create a new timesyncd.conf file in that directory to edit the file?

lrusak, 2021-11-16:
LibreELEC uses a squashfs root file system.

Please use the forum for support. https://forum.libreelec.tv/

closed 2021-11-16 by lrusak