LibreELEC/LibreELEC.tv #5880: [BUG]PCRE download site down

tinus-github, 2021-11-17:
### Describe the bug
Building the generic x86_64 build from scratch fails, because the downloads for the PCRE library have moved.

### To Reproduce
Steps to reproduce the behavior:
1. Download the 10.0 sources from git
2. Run PROJECT=Generic ARCH=x86_64 make image

Alternatively I guess you could remove everything in /sources/pcre and rebuild.

### Informations
 - LE Version: 10.0
 - Hardware Platform: generic x86_64

### Log file
If required I can see if I can find the build log and upload it.

### Additional context
As can be read on the site:

https://www.PCRE.org/

the old ftp.PCRE.org service is no longer available, so the package.mk file needs to be updated to point to another place. I found a mirror on osdn.net but it doesn’t have a neat URL.

SupervisedThinking, 2021-11-17:
Update your sources https://github.com/LibreELEC/LibreELEC.tv/commit/6d206bf1d518504e5728fe0abafdca75cb60c6d7

nomandera, 2021-11-17:
According to pcre.org the PCRE library is unmaintained

```
You can download the current release of the PCRE2 library from its official home on GitHub:

    https://github.com/PhilipHazel/pcre2/releases

You can also download PCRE2 or the older, unmaintained PCRE library from an unofficial mirror at SourceForge:

    https://sourceforge.net/projects/pcre/files/
```

tinus-github, 2021-11-17:
My build environment turned out to have been out of date.

closed 2021-11-17 by tinus-github