LibreELEC/LibreELEC.tv #5905: [Feature request] - Include dm-mod.ko by default.

ventoy, 2021-11-23:
### Describe
This is actually a feature request.
With [Ventoy](https://github.com/ventoy/Ventoy) we can just copy the LibreELEC.img to the USB drive and directly boot it.
It's very convenient for test and common use.
But Ventoy need the `dm-mod.ko` which is not included in the img file.
Can we build and contain the `dm-mod.ko` in the img file and load it by default in the future release?

PS:
I also made a request to `Lakka` and the developer has accepted it. 
Here is the link for your reference https://github.com/libretro/Lakka-LibreELEC/issues/1381

Thanks!

chewitt, 2022-11-28:
This forum report https://forum.libreelec.tv/thread/26320-problems-booting-img-files-from-10-x-version-using-ventoy-method/ states that Ventoy works with older LE 8.x/9.x releases, but not 10.x releases. As LE (and OE before) have never supported device-mapper modules in our kernel configs I conclude that something must have changed in Ventoy at some point that broke support? (and thus the fix should be in Ventoy to un-break whatever broke).

ventoy, 2022-11-28:
@chewitt 
Ventoy download the source code and build the dm.ko file for LE8.x/9.x.
That's why I made this feature request because I don't want to do this for every LE version in the future.

lrusak, 2022-11-28:
It's probably fine to include. I assume this only works for x86_64?

ventoy, 2022-11-29:
> It's probably fine to include. I assume this only works for x86_64?

x86_64 is mostly common used,  but Ventoy does support x86_64/i386/arm64/mips64el.

ventoy, 2022-11-29:
You can set `CONFIG_BLK_DEV_DM=y` in the dot.config for `i386, x86_64, and aarch64`

heitbaum, 2024-10-29:
Closing - info on forum. https://forum.libreelec.tv/thread/26320-problems-booting-img-files-from-10-x-version-using-ventoy-method/?postID=174370&highlight=ventoy#post174370

closed 2024-10-29 by heitbaum