LibreELEC/LibreELEC.tv #5918: Libreelec nightly show boot screen image when changing the resolution other than 1920x1080 on amd rx 6600 xt

e-ru, 2021-11-25:
### Describe the bug
The boot screen image shows up every time when changing the resolution other than 1920x1080.
This happens in the `System/Settings/Display/Resolution` Option and during video playback in the video options menu.

### To Reproduce
Steps to reproduce the behavior:
* Change resolution in System/Settings/Display/Resolution
or
* Change resolution in video options menu during video playback

### Informations
 - LE Version: nightly 20211024-73f3241
 - Hardware Platform: x86-64, cpu: intel 11400, gpu: rx 6600 xt

### Log file
http://ix.io/3G1U

### Additional context
-

lrusak, 2021-11-25:
Please report the bug to Kodi.

Your device is failing the atomic commit for some reason.

Please try and enable debug logging when posting a log.

Also post the output of the `modetest` command. Use `modetest | pastebinit`

e-ru, 2021-11-25:
ok thank you

i've mentioned this issue in the xmbc issue as well

regards
erik

heitbaum, 2022-02-17:
@e-ru  - is this still an issue? Is so can you please update this ticket with the current status and reopen this ticket.

closed 2022-02-17 by heitbaum