LibreELEC/LibreELEC.tv #5929: [BUG] Black screen with hdmi_enable_4kp60=1 on current nightly

dkadioglu, 2021-11-29:
### Describe the bug
Yesterday I updated to LE11 nightly from LE10 - wanted to test the new deinterlacer. After the update I had only a black screen. However, Kodi was running fine as the Android remote Yatse connected and worked as usual. Only when I commented hdmi_enable_4kp60=1 in /boot/config.txt I got my screen back.

### To Reproduce
Steps to reproduce the behavior:
1. Enable hdmi_enable_4kp60=1 in /boot/config.txt
2. Boot the Pi
3. Black Screen

### Informations
 - LE Version: 11 Nightly 20211128
 - Hardware Platform: RPi 4 8GB
 - Panasonic TV TX55-FXW784
 - Samsung HW-MS650 Soundbar
 - RPi HDMI0 (next to USB-C) is connected to HDMI1 of TV (for video) and HDMI1 is connected to Soundbar (for audio)

Is this known or currently expected? Because with LE 10 my setup works perfectly fine with hdmi_enable_4kp60=1.
What log output should I provide?

zehnerGIT, 2021-12-06:
Did you update from 10.0.0 to 11 or from 10.0.1?

If you have updated from 10.0.0, your problem sounds like the problem in https://forum.libreelec.tv/thread/24780-le-10-0-1-rpi4-no-picture-after-update-from-le-10-0-0/

If you have updated from 10.0.1, you've found a new way to let your HDMI output disappear ;-)

dkadioglu, 2021-12-07:
I've updated from a 10.0 self built build from 2021/10/15 from the 10.0 branch.
I also found out the following:
With hdmi_enable_4kp60=1 there is also no boot screen and no audio on HDMI 1. However, if I set Kodi UI to Full HD, at least I see the Kodi UI (still no audio though). Setting the resolution in Kodi to 4K does not work and resolution is restored to Full HD. As I had set the Kodi UI successfully to 4K when I used the 10.0 builds, after the update to the 11.0 build I had the black screen issue.

dkadioglu, 2021-12-20:
With a build from yesterdays master still the same: with hdmi_enable_4kp60=1 there is no boot screen, no hdmi audio on hdmi1 (I don't know if there is on hdmi0) and setting KODI UI to 4K is not possible. Is there anything of help, I could provide?

dkadioglu, 2022-01-08:
With a build from yesterdays master (00c51c726c15954625629788105840b0816f2688) still the same.

heitbaum, 2022-02-17:
@dkadioglu - given the number of changes and updates since %0- is this still a bug? If so please reopen.

closed 2022-02-17 by heitbaum

dkadioglu, 2022-05-19:
With a freshly installed nightly build from yesterday, this seems to be no issue anymore. I will observe and report back here if necessary.